jeremiah-c-leary/vhdl-style-guide #928: Unexpected token detected while parsing signal_declaration containing subtype

alexsh3, 2023-03-08:
**Environment**
Version: 3.14.0+zip.file
OS: Linux 4.18.0-372.26.1.el8_6.x86_64

**Describe the bug**
A signal declaration of a vector with a range defined by an integer subtype throws the following error:
```
Error: Unexpected token detected while parsing signal_declaration @ Line 10, Column 32 in file None
       Expecting : ;
       Found     : (
```

**To Reproduce**
Run `vsg -f ./range_failure.vhd` where `range_failure.vhd` contains the following:
```
library ieee;
  use ieee.std_logic_1164.all;

entity range_failure is
end entity range_failure;

architecture rtl of range_failure is

  subtype integer_range_subtype is integer range 3 downto 0;
  signal slv : std_logic_vector(integer_range_subtype);

begin

end architecture rtl;
```

**Expected behavior**
The code should parse correctly.

jeremiah-c-leary, 2023-04-01:
Morning @alexsh3,

I was able to duplicate the behavior you are seeing.  I will move this issue to In Progress.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-04-01:
Morning @alexsh3,

I have pushed an update for this issue to the `issue-928` branch.  When you get a chance, could you check it out and let me know if it fixes the issue on your side?

Thanks,

--Jeremy

alexsh3, 2023-04-03:
Hi @jeremiah-c-leary,

Branch `issue-928` has fixed this.

Thanks

Alex

jeremiah-c-leary, 2023-04-04:
Morning @alexsh3 ,

I will merge this to master.

Thanks,

--Jeremy

mattdominik, 2023-11-23:
we still have the same problem:

`library ieee; 
use ieee.std_logic_1164.all;

package interfaces_pkg is
    subtype T_AXI_RANGE is natural range 7 downto 0;

    type t_axi_stream_m2s is record
        st_valid : std_logic_vector(0 downto 0);
        st_data  : std_logic_vector;
        st_strb  : std_logic_vector;
        st_keep  : std_logic_vector;
        st_last  : std_logic_vector;
    end record;

    subtype t_data_axi_stream_m2s is t_axi_stream_m2s(
        st_data(7 downto 0),
        st_strb(0 downto 0),
        st_keep(T_AXI_RANGE),
        st_last(0 downto 0)
    );

end package;`

`Error: Unexpected token detected while parsing subtype_declaration @ Line 19, Column 36 in file None
       Expecting : ;
       Found     : ,`

jeremiah-c-leary, 2023-11-24:
Morning @mattdominik ,

I pushed an update to the `issue_928a` branch.  When you get a chance could you verify it on your end?

Thanks,

--Jeremy

mattdominik, 2023-11-28:
now it works - thanks

jeremiah-c-leary, 2023-11-29:
Awesome, I will merge this to master.

--Jeremy

PR #930 (2023-04-04): Issue#928:  Fixing classification of discrete_subtype_indication.
Resolves #928
PR #1051 (2023-11-29): Issue#928:  Subtype declaration classification issue
Resolves #928
closed 2023-11-29 by jeremiah-c-leary