jeremiah-c-leary/vhdl-style-guide #929: Rule process_032 cannot handle and incorrectly fixes postponed processes

JvWesterveld, 2023-03-31:
**Environment**
Installed via pip, `vsg --version` shows:

```
VHDL Style Guide (VSG) version: 3.14.0+zip.file
Git commit SHA: Unknown.  Installed via zip file.
```

OS: Ubuntu 22.04.

**Describe the bug**
It appears that a postponed keyword is interpreted as being the process label and is moved to the front of the line, which results in invalid VHDL. Disabling rule `process_032` fixes the issue.

**To Reproduce**
Create a VHDL file with the following content:

```vhdl
entity tb is
end entity tb;

architecture arch of tb is

begin

  label : process is
  begin

  end process label;

end architecture arch;
```

Run VSG with its default configuration:

`vsg -f filename.vhdl --fix`

Look at the resulting VHDL:
```vhdl
entity tb is
end entity tb;

architecture arch of tb is

begin
postponed label : process is
  begin

  end process label;

end architecture arch;
```

Note that the postponed keyword is now before the process label. When running VSG again you also get the following error:

```
Error: Unexpected token detected while parsing process_statement @ Line 7, Column 11 in file None
       Expecting : process
       Found     : label
```

**Expected behavior**
I would expect the postponed label to keep its relative position in between the label and the process keyword.

jeremiah-c-leary, 2023-04-08:
Afternoon @JvWesterveld ,

I pushed an update for this issue to the `issue-929` branch.  I replaced `process_032` with two rules:  `process_037` and `process_038`.  When you get a chance, let me know if it addresses the issue on your end.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-04-15:
Morning @JvWesterveld,

I added rule `process_039` to move force the postponed keyword to be on the same line as the the process keyword.  When you get a chance could you check it out on your end and let me know what you think?

Thanks,

--Jeremy

JvWesterveld, 2023-04-20:
Rule `process_039` seems to work well too, thanks!

jeremiah-c-leary, 2023-04-20:
Awesome,

I will get this merged to master.

--Jeremy

PR #933 (2023-04-20): Issue#929:  Fixing issue with moving label and the postponed keyword.
Resolved #929
closed 2023-04-20 by jeremiah-c-leary