jeremiah-c-leary/vhdl-style-guide #945: Check the case of the "return" keyword in functions

flyinged, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
Currently, there's no way to fix the case of the "return" keyword in a function definition. If the "return" keyword is written uppercase, it won't be fixed. 

**Describe the solution you'd like**
A rule that allows control on the case of the "return" keyword (case::keyword group)

**Describe alternatives you've considered**
I didn't find another way to control the case of the "return" keyword.

jeremiah-c-leary, 2023-10-11:
Good Evening @flyinged,

I pushed an update for this to the `issue-945` branch.  I added three rules: `function_501`, `return_statement_300` and `return_statement_500`.  When you get a chance could you check it out on your end and let me know if there are any issues.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-10-22:
Good Morning @flyinged ,

Just wanted to let you know I will be merging this into master.

--Jeremy

PR #1020 (2023-10-22): Issue-945:  Adding rules for return_statement production
Resolves #945
closed 2023-10-22 by jeremiah-c-leary