jeremiah-c-leary/vhdl-style-guide #946: Rule function_018 is documented but does not exist

flyinged, 2023-04-24:
**Environment**
Windows 10, WSL2, 3.14.0+zip.file

**Describe the bug**
Rule function_018 is documented but does not exist

**To Reproduce**
Run the following command:
`vsg -rc function_018`

Result:
`ERROR: rule function_018 was not found.`

The rule is actually not enforced: if `vsg `is run on code that matches the following pattern:

```
function func1 ( ... ) return ret_type is
begin
...
end;
```

The "end" statement won't be fixed, and no message will be returned from `vsg`.

**Expected behavior**
Configuration of rule should be printed

jeremiah-c-leary, 2023-04-24:
Morning @flyinged,

> Rule function_018 is documented but does not exist

Are you referring to the documentation on read-the-docs?

--Jeremy

flyinged, 2023-04-24:
Hi @jeremiah-c-leary 

> Are you referring to the documentation on read-the-docs?

Correct.
A.

JHertz5, 2023-07-27:
Hi @flyinged. You've mentioned in your original comment that you are using v3.14.0. The documentation for v3.14.0 does not include function_018.

It looks like function_018 was added after that version. I am running v3.16.0 and `vsg -rc function_018` prints out the configuration as expected.

I suggest that you either update the version of vsg, or use the documentation that corresponds to the version that you are using. You can modify the version of the documentation in the bottom left of the browser window on readthedocs.io

jeremiah-c-leary, 2023-08-06:
Evening @flyinged ,

You can pick the version of documentation by clicking on the following at the bottom left of the read-the-docs window:

[image]

Let me know if this addresses your question. 

--Jeremy

flyinged, 2023-08-14:
Ok. Problem solved. Thanks