jeremiah-c-leary/vhdl-style-guide #947: Rule generate_015 fix can turn valid code into invalid code

flyinged, 2023-04-24:
**Environment**
Windows 10, WSL2, 3.14.0+zip.file

**Describe the bug**
Rule generate_015 fix can turn valid code into invalid code in certain situations (s. example below)

**To Reproduce**
Run `vsg ` on code matching the following pattern:
```
label1 :  -- some comment
  case param1 generate
  when 1 =>
  ...
  end generate label1;
```

The fixed code will be:
```
label1 :  -- some comment  case param1 generate
  when 1 =>
  ...
  end generate label1;
```

The original code is legal according to VHDL-2008 standard, but the fixed code is not.

**Expected behavior**
The comment after the label should be either removed or moved one line above.

jeremiah-c-leary, 2023-05-10:
Evening @flyinged,

I pushed an update to this to the `issue-947` branch.  When you get a chance could you run it on your end and let me know if the issue is resolved on your end?

Thanks,

--Jeremy

flyinged, 2023-05-15:
Hi @jeremiah-c-leary 

I tried the `issue-947` branch against the following code:

```
library IEEE;
use IEEE.std_logic_1164.all;

entity test is
port(
	clk : in std_logic;
	din : in std_logic;
	dout : out std_logic
);
end entity;

architecture behav of test is

begin

label1 : --some comment
	case param1 generate
	when 1 =>
		--do something
	when others =>
		--do something else
end generate label1;

end architecture;
```

and I obtained the following "fixed" code:
```
library ieee;
  use ieee.std_logic_1164.all;

entity test is
  port (
    clk  : in    std_logic;
    din  : in    std_logic;
    dout : out   std_logic
  );
end entity test;

architecture behav of test is

begin

  label1 : -- some comment  case param1 generate
    when 1 =>
    -- do something
    when others =>
  -- do something else
  end generate label1;

end architecture behav;
```

So I would say that the problem ist still there.

A.

jeremiah-c-leary, 2023-08-09:
Morning @flyinged ,

I just ran your above example on the `issue-947` branch and I get the following fixed version:
```vhdl
  1 library ieee;
  2   use ieee.std_logic_1164.all;
  3
  4 entity test is
  5   port (
  6     clk  : in    std_logic;
  7     din  : in    std_logic;
  8     dout : out   std_logic
  9   );
 10 end entity test;
 11
 12 architecture behav of test is
 13
 14 begin
 15
 16   -- some comment
 17
 18   label1 : case param1 generate
 19     when 1 =>
 20     -- do something
 21     when others =>
 22   -- do something else
 23   end generate label1;
 24
 25 end architecture behav;
```
Could you double check you are using the `issue-947` branch?

Thanks,

--Jeremy

flyinged, 2023-08-14:
Hi @jeremiah-c-leary 

I double checked. On branch issue-947 it works as you described. 

Thanks
A.

PR #955 (2024-03-10): Issue#947:  Rule generate_015 could produce invalid code.
Resolves #947
closed 2023-08-14 by flyinged