jeremiah-c-leary/vhdl-style-guide #949: Is there a way to enforce the case of the "return" keyword?

flyinged, 2023-04-25:
**What is your question?**
Is there a way to enforce the case of the `return` keyword in a function declaration and in its body?
With the current default settings, an uppercase `return` in either a function definition or a function body won't trigger any rule.

jeremiah-c-leary, 2023-04-25:
Morning @flyinged,

Adding a case rule for the return keyword is pretty easy.

--Jeremy

flyinged, 2023-04-25:
> Morning @flyinged,
> 
> Adding a case rule for the return keyword is pretty easy.
> 
> --Jeremy

Hi @jeremiah-c-leary 

Until now, I just used the existing rules. Is there some kind of example one can start with?

Alessandro

flyinged, 2023-05-08:
Hi @jeremiah-c-leary 

I've tried to follow your instructions:
- created rule_503.py in folder `vsg/rules/function` based on `vsg/rules/reports_statement/rule_500.py`
- updated `__init__.py`
- created `test_rule_503.py` and the relative VHDL files

Command `python -m unittest vsg.tests.function.test_rule_503` is successful.
Command `python -m unittest` fails with several errors (as expected).

Then I made the following changes:

- Updated `docs/function_rules.rst`
- Updated `docs/rule_groups/case_keyword_rule_group.rst` 
- Updated `docs/rule_groups/case_rule_group.rst` 

I still get some errors:

```
.............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................F................FFF........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
======================================================================
FAIL: test_rule_link_in_configuration_documentation_exists (vsg.tests.rule_doc.test_rule_doc.testDocGen)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/alessandro/vhdl-style-guide/vsg/tests/rule_doc/test_rule_doc.py", line 83, in test_rule_link_in_configuration_documentation_exists
    self.assertEqual(dConfigurationFiles[sKey], lActual)
AssertionError: Lists differ: ['ali[1532 chars]on_503', 'function_506', 'generate_005', 'gene[1756 chars]004'] != ['ali[1532 chars]on_506', 'generate_005', 'generate_009', 'gene[1740 chars]004']

First differing element 77:
'function_503'
'function_506'

First list contains 1 additional elements.
First extra element 177:
'variable_004'

  ['alias_declaration_500',
   'alias_declaration_501',
   'alias_declaration_502',
   'architecture_004',
   'architecture_009',
   'architecture_011',
   'architecture_013',
   'architecture_014',
   'architecture_019',
   'architecture_020',
   'architecture_021',
   'architecture_028',
   'attribute_500',
   'attribute_declaration_500',
   'attribute_declaration_501',
   'attribute_declaration_502',
   'attribute_specification_500',
   'attribute_specification_501',
   'attribute_specification_502',
   'attribute_specification_503',
   'block_500',
   'block_501',
   'block_502',
   'block_503',
   'block_504',
   'block_505',
   'block_506',
   'case_014',
   'case_015',
   'case_016',
   'case_017',
   'case_018',
   'case_generate_alternative_500',
   'case_generate_alternative_501',
[... 173 lines cut ...]
======================================================================
FAIL: test_config_3 (vsg.tests.rule_group.test_rule_group.testMain)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/alessandro/vhdl-style-guide/vsg/tests/rule_group/test_rule_group.py", line 124, in test_config_3
    self.assertEqual(lExpected, lActual)
AssertionError: Lists differ: ['', [1057 chars]NC_1 Return integer;', '', 'end package SOME_P[147 chars]DY;'] != ['', [1057 chars]NC_1 return integer;', '', 'end package SOME_P[147 chars]DY;']

First differing element 57:
'  function FUNC_1 Return integer;'
'  function FUNC_1 return integer;'

Diff is 1557 characters long. Set self.maxDiff to None to see it.

----------------------------------------------------------------------
Ran 2585 tests in 53.460s

FAILED (failures=4)
```

Any suggestions on how to solve it? The error messages are not so informative.

Also: why do I get the message `ERROR: rule function_503 was not found` when I run `vsg -rc function_503`? How can I define the rule's default configuration?

A.

flyinged, 2023-05-15:
Hi @jeremiah-c-leary 

Thanks for the suggestions.

> The first error is checking the documentation for a link to the rule you just added. I agree it is completely unclear on how to fix it. Try this: grep for `function_506` in all the documentation. You are looking for either configuration documentation or in the rule_group directory. Try to match the array to the documentation and see where `function_503` should be.

This was solved by updating `docs/configuring_uppercase_and_lowercase_rules.rst`

> The next three are checking that `rule_group`s work. Since you added a rule to check the case of the return keyword, the tests are now failing.

These could also be solved by following your hints.

> The `-rc` function works as long as the rule is available. Double check if you are calling the correct `vsg`.

This still does not work. I call `vsg` from the repository'y root with `./bin/vsg -rc function_503`. Should I call `vsg` in another way?

A.

jeremiah-c-leary, 2023-10-27:
Hey @flyinged ,

I am going to close this issue on November 1st, as I believe issue #945 has implemented a fix for this issue.  If you would like to keep this issue open, please let me know.

Regards,

--Jeremy

flyinged, 2023-10-28:
Hi Jeremy

Sorry for the missed answers. I had a busy today me. Feel free to close the
issue if you think it's solved

Regards
Alessandro

jeremiah-c-leary, 2023-10-28:
Hey @flyinged ,

No problem.  I will close this issue.

--Jeremy

closed 2023-10-28 by jeremiah-c-leary