jeremiah-c-leary/vhdl-style-guide #950: procedure_012 error with function inside procedure

avelure, 2023-04-25:
**Environment**
V3.15.0, the error is not present in 3.14.0 but I see the check was recently introduced

**Describe the bug**
A procedure that has a function in the declarative part will give procedure_012 error on the end keyword of the function.

**To Reproduce**
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
begin
  b_test : block is
    procedure p_test (
      signal   val3 : inout signed;
      signal   val4 : inout signed
    ) is
      function f_test (
        constant val1 : std_logic;
        constant val2 : std_logic
      ) return signed is
      begin
      end function f_test;
    begin
    end procedure p_test;
  begin
  end block b_test;
end architecture beh;
```
`vsg -f procedure_012.vhd`

```
================================================================================
File:  procedure_012.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 157
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  procedure_012             | Error      |         19 | Add procedure keyword
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
No error.

jeremiah-c-leary, 2023-10-27:
Good Morning @avelure ,

I pushed an update to this to the `issue-1022` branch.  When you get a chance could you check it out on your end and let me know how it goes?

Thanks,

--Jeremy

avelure, 2023-10-30:
Hi,
thanks for the update.
I can confirm it works now,

jeremiah-c-leary, 2023-10-31:
Morning @avelure ,

Awesome, I will get this merged into master.

--Jeremy

PR #1028 (2023-11-02): Issue#1022:  Fixing rule function_018.
  * Rule could not determine the difference between subprogram_body and subprogram_declaration.

Resolves #1022 
Resolves #950
closed 2023-11-02 by jeremiah-c-leary