jeremiah-c-leary/vhdl-style-guide #951: Port suffix rules 605, 606, 607 fail on blank lines

alexsh3, 2023-04-28:
**Environment**
VHDL Style Guide (VSG) version: 3.15.0+zip.file
OS: Linux 4.18.0-425.13.1.el8_7.x86_64

**Describe the bug**
When running rules port_605, port_606, and port_607 on an entity with blank lines in the port map, the following is reported:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  port_605                  | Error      |          9 | Suffix  with one of the following: _i
  port_606                  | Error      |         13 | Suffix  with one of the following: _o
  port_607                  | Error      |         17 | Suffix  with one of the following: _io
```

**To Reproduce**
Run on the below file:
```
library ieee;
  use ieee.std_logic_1164.all;

entity port_suffix_test is
  port (
library ieee;
  use ieee.std_logic_1164.all;

entity port_suffix_test is
  port (
    -- line 6
    port_0_i : in    std_logic;
    port_1_i : in    std_logic;

    -- line 10
    port_2_i : in    std_logic;
    port_3_i : in    std_logic;

    -- line 14
    port_4_o : out   std_logic_vector(31 downto 0);
    port_5_o : out   std_logic_vector(31 downto 0);

    -- line 18
    port_6_io : inout std_logic_vector(31 downto 0);
    port_7_i  : in    std_logic_vector(31 downto 0);
    port_8_i  : in    std_logic_vector(31 downto 0);
    port_9_o  : out   std_logic_vector(31 downto 0)
  );
end entity port_suffix_test;
  );
end entity port_suffix_test;

```
The rule configuration is as below:
```
{
  "rule": {
    "port_605": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 7,
      "disable": false,
      "fixable": false,
      "severity": "Error",
      "suffixes": [
        "_i"
      ]
    },
    "port_606": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 7,
      "disable": false,
      "fixable": false,
      "severity": "Error",
      "suffixes": [
        "_o"
      ]
    },
    "port_607": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 7,
      "disable": false,
      "fixable": false,
      "severity": "Error",
      "suffixes": [
        "_io"
      ]
    }
  }
}
```

**Expected behavior**
I would expect these rules to ignore blank lines.

jeremiah-c-leary, 2023-05-04:
Evening @alexsh3 ,

I pushed a fix for this to the `issue-951` branch.  When you get a chance could you check it out on your end and let me know if it works for you.

Thanks,

--Jeremy

alexsh3, 2023-05-09:
Hi @jeremiah-c-leary,

It looks like the fix in `issue-951` resolves this issue.

Thanks,

Alex

PR #954 (2023-05-10): Issue#951:  Fixing suffix checks on blank lines.
Resolves #951
closed 2023-05-10 by jeremiah-c-leary