jeremiah-c-leary/vhdl-style-guide #952: Case consistency for usage of enumerated types

alexsh3, 2023-04-28:
When running rule `type_500` configured as below on the attached example code, the rule correctly produces the following output for the type declaration:
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  type_500                  | Error      |         11 | Change "s_enum_1" to "S_ENUM_1"
  type_500                  | Error      |         11 | Change "S_enum_3" to "S_ENUM_3"
  type_500                  | Error      |         11 | Change "s_enum_4" to "S_ENUM_4"
```
As far as I can see, though, there is no way to enforce a consistent case when this type is  used, as on lines 21-24.

It would be really useful to have a rule that could enforce case consistency.

I apologise if I have missed an existing way of accomplishing this.

Thanks,

Alex

### Example test code
```
library ieee;
  use ieee.std_logic_1164.all;

entity enum_type_test is
end entity enum_type_test;

architecture test of enum_type_test is

  -- line 9

  type t_enum is (s_enum_1, S_ENUM_2, S_enum_3, s_enum_4);

  signal sig_1 : t_enum;
  signal sig_2 : t_enum;
  signal sig_3 : t_enum;
  signal sig_4 : t_enum;

begin

  -- line 20
  sig_1 <= s_enum_1;
  sig_2 <= S_ENUM_2;
  sig_3 <= S_enum_3;
  sig_4 <= s_enUM_4;

end architecture test;

```
### Rule config
```
{
  "rule": {
    "type_500": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 6,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "case": "upper"
    }
  }
}
```

jeremiah-c-leary, 2023-05-10:
Evening @alexsh3 ,

I added a new rule `type_501` which should enforce consistent case for enumerated types.  Could you check it out on your end and let me know what you think.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-08-06:
Evening @alexsh3 ,

I am going to merge this to master.

Regards,

--Jeremy

PR #956 (2023-08-06): Issue#952: Adding consistent case check for enumerated types.
Resolves #952
closed 2023-08-06 by jeremiah-c-leary