jeremiah-c-leary/vhdl-style-guide #959: port_605/port_606 rules not working properly with doxygen-commented entity

abyszuk, 2023-06-06:
**Environment**
OS: Ubuntu 22.04
Python version: 3.10
VSG version: 3.15.0, installed via `pip`

**Describe the bug**
With the entity below rules port_605 and port_606 don't seem to work correctly.
VSG reports errors on empty lines, and even if they were to apply to the previous line they are wrong.

**To Reproduce**
Use this MWE:
```vhdl
library ieee;
use ieee.std_logic_1164.all;

--! Top integration module for PB
entity pb_top is
  port (
    --! @name Basic input signals
    mgt_clk_p_i      : in std_logic; --! Aurora MGT clock
    mgt_clk_n_i      : in std_logic;
    hlp_dcxo_clk_p_i : in std_logic; --! Helper DCXO clock
    hlp_dcxo_clk_n_i : in std_logic; --! Helper DCXO clock

    --! @name DIOT backplane basic signals
    bp_ga_i          : in std_logic_vector(3 downto 0); --! Geographical address
    bp_sys_servmod_i : in std_logic; --! SERVMOD state
    bp_lvds_p_i      : in std_logic_vector(14 downto 0); --! LVDS IO (unused)
    bp_lvds_n_i      : in std_logic_vector(14 downto 0);
    bp_sh_i          : in std_logic_vector(4 downto 0); --! Shared IO

    -- @name Backplane MGT
    mgt_tx_p_o : out std_logic; --! Aurora TX
    mgt_tx_n_o : out std_logic;
    mgt_rx_p_i : in std_logic; --! Aurora RX
    mgt_rx_n_i : in std_logic;

    --! @name Development IO
    uart_rx_i : in std_logic; --! UART RX
    uart_tx_o : out std_logic; --! UART TX
    led_o     : out std_logic_vector(3 downto 0) --! User LEDs
  --! @}
  );
end pb_top;

--! Architecture implementation for PB top entity
architecture arch of pb_top is

begin

end arch;
```
with this config:
```yaml
  port_605:
    disable: false
    fixable: false
    indentSize: 2
    indentStyle: spaces
    phase: 7
    severity: Error
    suffixes:
    - _i
  port_606:
    disable: false
    fixable: false
    indentSize: 2
    indentStyle: spaces
    phase: 7
    severity: Error
    suffixes:
    - _o
 ```

**Expected behavior**
MWE entity is parsed properly.

jeremiah-c-leary, 2023-06-10:
Morning @abyszuk ,

I pushed an update for this issue to the `issue-960` branch.  When you get a chance could you check it on your end and validate it addresses the problem.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-06-10:
Morning @abyszuk ,

It looks like this was fixed with issue #951 and is scheduled for the next release.  I will do another release once issue #960 has been validated.

Regards,

--Jeremy

closed 2023-06-10 by jeremiah-c-leary