jeremiah-c-leary/vhdl-style-guide #960: VSG crashes when incorrect code tags are applied

abyszuk, 2023-06-06:
**Environment**
OS: Ubuntu 22.04
Python: 3.10
VSG: 3.15.0 installed via `pip`

**Describe the bug**
With the following `mwe.vhd` and configuration file VSG crashes.
There's a minor mistake in that file - the order of `vsg_off` and `vsg_on` tags is reversed, but IMHO that shouldn't lead to a crash.
```vhdl
library ieee;
use ieee.std_logic_1164.all;

--! @brief Helper package for module aurora_c2c
package ex_pkg is

  -- vsg_on port_025 port_605 port_606
  --! Xilinx Aurora XCI IP declaration (watch out for inverted AXI slv range!)
  component aurora
    port (
      s_axi_tx_tdata  : in std_logic_vector(0 to 31);
      s_axi_tx_tvalid : in std_logic;
      s_axi_tx_tready : out std_logic;
      m_axi_rx_tdata  : out std_logic_vector(0 to 31);
      m_axi_rx_tvalid : out std_logic;

      rxp                : in std_logic;
      rxn                : in std_logic;
      txp                : out std_logic;
      txn                : out std_logic;
      gt_refclk1         : in std_logic;
      hard_err           : out std_logic;
      soft_err           : out std_logic;
      lane_up            : out std_logic;
      channel_up         : out std_logic;
      user_clk_out       : out std_logic;
      sync_clk_out       : out std_logic;
      gt_reset           : in std_logic;
      reset              : in std_logic;
      sys_reset_out      : out std_logic;
      gt_reset_out       : out std_logic;
      power_down         : in std_logic;
      loopback           : in std_logic_vector( 2 downto 0);
      tx_lock            : out std_logic;
      init_clk_in        : in std_logic;
      tx_resetdone_out   : out std_logic;
      rx_resetdone_out   : out std_logic;
      link_reset_out     : out std_logic;
      gt0_drpaddr        : in std_logic_vector( 8 downto 0);
      gt0_drpdi          : in std_logic_vector( 15 downto 0);
      gt0_drpdo          : out std_logic_vector( 15 downto 0);
      gt0_drpen          : in std_logic;
      gt0_drprdy         : out std_logic;
      gt0_drpwe          : in std_logic;
      gt_powergood       : out std_logic_vector( 0 to 0);
      pll_not_locked_out : out std_logic
    );
  end component aurora;
-- vsg_off

end package ex_pkg;
```
Crash log:
```
adrian@pcte247806:~/praca/cce/fgc4$ vsg -c vsg_rules.yml -f mwe.vhd
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/apply_rules.py", line 67, in apply_rules
    oVhdlFile = vhdlFile.vhdlFile(lFileContent, sFileName, eError)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/vhdlFile.py", line 69, in __init__
    self._processFile()
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/vhdlFile.py", line 101, in _processFile
    set_code_tags(self.lAllObjects)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/vhdlFile.py", line 582, in set_code_tags
    oCodeTags.update(oToken)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/code_tags.py", line 42, in update
    remove_code_tags(self, oToken)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/code_tags.py", line 89, in remove_code_tags
    self.remove(sCodeTag)
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/vhdlFile/code_tags.py", line 20, in remove
    self.code_tags.remove(sCodeTag)
ValueError: list.remove(x): x not in list
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/adrian/.local/bin/vsg", line 8, in <module>
    sys.exit(main())
  File "/home/adrian/.local/lib/python3.10/site-packages/vsg/__main__.py", line 142, in main
    for tResult in pool.imap(f, enumerate(commandLineArguments.filename)):
  File "/usr/lib/python3.10/multiprocessing/pool.py", line 873, in next
    raise value
ValueError: list.remove(x): x not in list
```

**To Reproduce**
Unpack [mwe.zip](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/11665354/mwe.zip) and run `vsg -c vsg_rules.yml -f mwe.vhd`

jeremiah-c-leary, 2023-06-10:
Morning @abyszuk ,

I pushed an update to this to the `issue-960` branch.  When you get a chance could you validate it addresses the problem on your end.

Thanks,

--Jeremy

abyszuk, 2023-07-05:
Sorry for the late reply, I was on vacation.

I confirm that this works now. Thank you!

jeremiah-c-leary, 2023-07-24:
Evening @abyszuk ,

Awesome, I will get this merged to master.

--Jeremy

PR #962 (2023-07-24): Issue#960:  Fixing crash if code tag enabled when it was not previous…
Resolves #960
closed 2023-07-24 by jeremiah-c-leary