jeremiah-c-leary/vhdl-style-guide #966: Unexpected token detected while parsing subtype_declaration containing subtype

mkuklewski, 2023-07-04:
**Environment**
Version: 3.15.0+zip.file
OS: Windows 10

**Describe the bug**
Following code throws the error:

```
 Error: Unexpected token detected while parsing signal_declaration @ Line 10, Column 32 in file None
        Expecting : ;
        Found     : (
```

Similar issue to this #928

**To Reproduce**
To Reproduce
Run vsg -f ./subtype_failure.vhd where subtype_failure.vhd contains the following:

```
library ieee;
  use ieee.std_logic_1164.all;

entity subtype_failure is
end entity subtype_failure;

architecture rtl of subtype_failure is

  subtype Data_r is natural range 8 - 1 downto 0;

  type SlaveIn_t is record
    RdDataxD            : std_logic_vector;
  end record SlaveIn_t;

  subtype SlaveInCnstrd_t is SlaveIn_t(
    RdDataxD(Data_r)
  );

begin

end architecture rtl;
```

**Expected behavior**
The code should parse correctly.

jeremiah-c-leary, 2023-07-15:
Morning @mkuklewski ,

I was able to recreate the above error.  I will take a look at the parsing.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-10-28:
Morning @mkuklewski ,

Better late than never?

I pushed an update for this issue to the `issue-966a` branch.  When you get a chance could you run it on your end and let me now how it goes?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-11-19:
Morning @mkuklewski ,

I am going to merge this to master.  If there is still a problem with the implementation, please create another issue.

Thanks,

--Jeremy

PR #1032 (2023-11-19): Issue#966: Updating classifier to handle subtypes within constraints.
Resolves #966
closed 2023-11-19 by jeremiah-c-leary