jeremiah-c-leary/vhdl-style-guide #967: Signal_012 aligned to wrong group

t-aras, 2023-07-11:
**Environment**
Version: 3.15.0
Linux

**Describe the bug**
The `signal_012` rule does align over all signal declarations. An empty line or a comment line does not end the group.
Therefore the second in-line signal declarations is moved too far left.

**To Reproduce**
File to reproduce:
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity test is
  port (
    RstxRBI : in    std_logic;
    ClkxCI  : in    std_logic;
    TestxCI : out   std_logic
  );
end entity test;

architecture rtl of test is

  signal Test1P, Test1N         : std_logic;
  signal Test2                  : std_logic_vector(32 - 1 downto 0);
  signal RdRdyN, RdRdyP         : std_logic;

  signal ExtrLongSignalName1, ExtrLongSignalName1 : std_logic;

begin

end architecture rtl;
```

After running the fix for signal_012 the file looks like that:
```
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity test is
  port (
    RstxRBI : in    std_logic;
    ClkxCI  : in    std_logic;
    TestxCI : out   std_logic
  );
end entity test;

architecture rtl of test is

  signal Test1P,              Test1N : std_logic;
  signal Test2                       : std_logic_vector(32 - 1 downto 0);
  signal RdRdyN,              RdRdyP : std_logic;

  signal ExtrLongSignalName1, ExtrLongSignalName1 : std_logic;

begin

end architecture rtl;
```

**Expected behavior**
I would expect that the file should not be touched.

jeremiah-c-leary, 2023-07-15:
Morning @t-aras ,

I agree with your expected behavior.  I will add the comment and blank line options seen in other alignment rules.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-11-19:
Good Afternoon @t-aras ,

I added the following two options to the rule:  `blank_line_ends_group` and `comment_line_ends_group`.

I pushed the update to the `issue-967` branch.  When you get a chance could you check it out and let me know what you think.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @t-aras ,

Just wanted to ping you on this issue to see if you had time to check it out.

Thanks,

--Jeremy

t-aras, 2023-12-15:
Hey Jeremy

Sorry, didn't had the time yet to test it. I am on vacation and will only be able to test it in the new year. Same for the other open issue with a fix ready: https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/998
Thank you for the fix.

I'll give you feedback as soon as I have tested it.

jeremiah-c-leary, 2024-01-21:
Afternoon @t-aras ,

Wondering if you have time to check this out yet.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-02-04:
Morning @t-aras,

Just a ping to see if you had a chance to check this out yet.

I will plan to merge this to master by 2/18 if I do not here back from you.

Thanks,

--Jeremy

t-aras, 2024-02-04:
Hi @jeremiah-c-leary 

Just tested it and it works fine on my end.

Thank you for the fix!

Best
@t-aras

jeremiah-c-leary, 2024-02-04:
Awesome,

I will get it merged to master.

--Jeremy

PR #1050 (2024-02-04): Issue#967:  Adding options to rule signal_012
   * Added blank_line_ends_group
   * Added comment_line_ends_group

Resolves #967
closed 2024-02-04 by jeremiah-c-leary