jeremiah-c-leary/vhdl-style-guide #969: Library_007 exception for the work library

cla5323, 2023-07-13:
Currently, the `library_007` rule will remove blank lines before use clauses. This normally isn't a problem since you usually include the library before pulling in the desired modules/packages from that library so `library_003` takes effect. However, this becomes a problem with the work library since it's included by default and thus doesn't need the `library work;` statement. This results in use clauses from the work library to immediately proceed those from another library.
</br>
My recommendations for this are to either:
1. Add an exception for use clauses from the work library
2. Add an option similar to `allow_library_clause` on `library_003` that would allow consecutive use clauses from different libraries
</br>
Example code:

```vhdl
library ieee;

    use ieee.numeric_std.all;

    use work.registers_pkg.all;

    use work.utility_pkg.all;
```

</br>
Current behavior:

```vhdl
library ieee;
    use ieee.numeric_std.all;
    use work.registers_pkg.all;
    use work.utility_pkg.all;
```

</br>
Desired behavior:

```vhdl
library ieee;
    use ieee.numeric_std.all;

    use work.registers_pkg.all;
    use work.utility_pkg.all;
```

jeremiah-c-leary, 2023-07-15:
Morning @cla5323 ,

I agree the libraries should be separated.   It seems we would want the the desired behavior to be the default behavior.

Here is a question:  Should the work use clauses be left aligned?
```vhdl
library ieee;
    use ieee.numeric_std.all;

use work.registers_pkg.all;
use work.utility_pkg.all;
```
Given there is no library clause for work.

Regards,

--Jeremy

cla5323, 2023-07-17:
Hey Jeremy,

I like to keep the work use clauses aligned with the rest of them since they are use clauses, but I can see an argument for wanting them left aligned. Maybe it should be a rule/option that the user can select?

- Chris

JHertz5, 2023-07-25:
Just to put it another voice - I would love to see this implemented and would personally like to use it as shown in Jeremy's example with the use cases left-aligned, so if it's not too much work, an option would be wonderful.

jeremiah-c-leary, 2023-10-23:
Good Evening @cla5323  and @JHertz5 ,

I pushed an update for this issue to the `issue-969` branch.  When you get a chance could you check it out and let me know what you think.

I have not addressed the left alignment of work libraries yet.

Regards,

--Jeremy

JHertz5, 2023-10-23:
Hi @jeremiah-c-leary. That branch looks to be working on my end. As mentioned previously, if left-alignment is a pain, I am totally happy for it to be ignored or I can raise a separate issue.

cla5323, 2023-10-24:
Hi @jeremiah-c-leary. I also gave it a try and things look good to me.

Thanks,

- Chris

jeremiah-c-leary, 2023-10-25:
Morning @JHertz5 and @cla5323 ,

Awesome.  @JHertz5 , could you create a new issue for aligning the work library.  There could be some interesting discussion on indenting with respect to context clauses.

I will merge this to master after the new issue has been created.

Thanks,

--Jeremy

JHertz5, 2023-10-25:
@jeremiah-c-leary Of course! I have created #1030.

PR #1025 (2023-10-26): Issue#969:  Updating use_clause rules for more user control.
Resolves #969
closed 2023-10-26 by jeremiah-c-leary