jeremiah-c-leary/vhdl-style-guide #975: generate_011

JHertz5, 2023-07-25:
**Environment**
v3.16.0

**Describe the bug**
The rule `generate_011` configures an optional VHDL item, and is part of the structure::optional group. However, unlike the other rules in that group, it does not have an `action` configuration like the [other optional item rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_optional_items.html#configuring-optional-items). This means that, even though `generate_011` is part of this group, it isn't affected by setting this `action` configuration for the group.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `vsg -rc generate_011`
2. See that output does not include the `action: '[add|remove]'` option

**Expected behavior**
I expect generate_011 to have this configuration item so that it can be configured along with the rest of the group.

**Screenshots**
None

**Additional context**
None

jeremiah-c-leary, 2023-08-06:
Afternoon @JHertz5 ,

I pushed an update for this issue to the `issue-975` branch.  When you get a chance could you validate it works on your end.

Thanks,

--Jeremy

JHertz5, 2023-08-07:
Hi @jeremiah-c-leary. Thank you so much! I have taken a look. It is working excellently, except for the following error message when the rule is configured to remove the label:
```
================================================================================
File:  test.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 157
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  generate_011              | Error      |          7 | Remove Add generate label
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
The solution message says "Remove Add generate label, where I would expect it to only say "Remove generate label".

I have added a review to the pull request to suggest a fix, I hope that is ok.

JHertz5, 2023-08-07:
@jeremiah-c-leary Thanks, the update has resolved that issue. It all looks good to me!

jeremiah-c-leary, 2023-08-07:
Sweet,

I will merge this to master.

--Jeremy

PR #989 (2023-08-07): Issue#975:  Adding "remove" and "add" options to rule generate_011.
Resolves #975
closed 2023-08-07 by jeremiah-c-leary