jeremiah-c-leary/vhdl-style-guide #976: Support prefix for functions

javValverde, 2023-07-27:
**Is your feature request related to a problem? Please describe.**
I would like to have a rule that checks the prefix for functions

**Describe the solution you'd like**
I would do it similar rules such as `constant_015`

**Describe alternatives you've considered**
None

**Additional context**
None

jeremiah-c-leary, 2023-08-06:
Afternoon @javValverde ,

I will add this to the backlog.

--Jeremy

jeremiah-c-leary, 2023-08-06:
Afternoon @javValverde ,

I push an update for this issue to the issue-976 branch.

I added rules `function_600` and `function_601`.

When you get a chance could you check if it working on your end.

Thanks,

--Jeremy

javValverde, 2023-08-07:
I tested `function_600` and it is working as intended.
Thanks you!

jeremiah-c-leary, 2023-08-07:
Awesome, I will merge this to master.

--Jeremy

PR #987 (2023-08-07): Issue#976:  Added prefix and suffix rules for function designators.
Resolves #976
closed 2023-08-07 by jeremiah-c-leary