jeremiah-c-leary/vhdl-style-guide #977: Incorrect name and broken link when referencing case_generate_statement_400 in the Configuring Disabled Rules page

JHertz5, 2023-07-28:
**Environment**
v3.16.0

**Describe the bug**
The page https://vhdl-style-guide.readthedocs.io/en/latest/configuring_disabled_rules.html lists `case_generate_statement_300` as one of the rules but links to case-statement-generate-400. Additionally, the link is broken; it directs the user to case_generate_statement.html#case-statement-generate-400 rather than case_generate_statement_rules.html#case-generate-statement-400

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '[Configuring Disabled Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_disabled_rules.html)'
2. Click on 'case_generate_statement_300'
3. See error

**Expected behavior**
This item should list, and correctly link to [case-statement-generate-400](https://vhdl-style-guide.readthedocs.io/en/latest/case_generate_statement_rules.html#case-generate-statement-400).

**Screenshots**
[image]

**Additional context**
None

jeremiah-c-leary, 2023-08-06:
Morning @JHertz5 ,

I have pushed an update to master for the broken link.  Could you could validate the link works on read-the-docs?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-08-06:
Morning @JHertz5,

I just realized you had a pull request that covered this.

I will close this issue.

JHertz5, 2023-08-06:
Hi @jeremiah-c-leary, thanks very much! I have checked read-the-docs and can confirm that the issue is resolved. I agree that you can close the issue. Thanks again!

closed 2023-08-06 by jeremiah-c-leary