jeremiah-c-leary/vhdl-style-guide #981: Block comments YAML example should have blank spaces instead of "None".

JHertz5, 2023-07-28:
**Environment**
v3.16.0

**Describe the bug**
https://vhdl-style-guide.readthedocs.io/en/latest/configuring_block_comments.html#configuring-block-comments gives examples of how to configure block comments with YAML configuration files. However, where the rule is expected to enforce a "empty string", the example uses "None". This results in the rule expecting "NoneNoneNone..." rather than the empty string as intended.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://vhdl-style-guide.readthedocs.io/en/latest/configuring_block_comments.html#configuring-block-comments
2. Copy the Simple Block Comment rule configuration into a YAML file.
3. Run vsg with this configuration on a file containing a block comment.
4. See 
```
  block_comment_001         | Error      |         62 | Change block comment header to : --None---------------------------------------------------NoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNone
  block_comment_002         | Error      |         63 | Comment must start with --None#
  block_comment_003         | Error      |         64 | Change block comment footer to : --None---------------------------------------------------NoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNoneNone
```

**Expected behavior**
I expect the example to configure empty strings instead of "None" in these instances.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

JHertz5, 2023-07-28:
I could correct this by changing the YAML examples to have an empty string instead of "None", but the JSON uses "null" instead of an empty string here. Anyone who follows the example for JSON by trying to use an empty string may run into problems. Should this page give examples of both YAML and JSON so that the user knows what to do in both cases?

jeremiah-c-leary, 2023-08-06:
Morning @JHertz5 ,

I followed your instructions and was able to replicate the `None` being shown in the solution.

I like your idea of switching from `None` to and empty string `''`.

```yaml
  1
  2 rule:
  3   block_comment_001:
  4     disable : False
  5     header_left : '-'
  6     header_left_repeat : '-'
  7     header_string : ''
  8     header_right_repeat : ''
  9   block_comment_002:
 10     disable : False
 11     comment_left : '!'
 12   block_comment_003:
 13     disable : False
 14     footer_left : '-'
 15     footer_left_repeat : '-'
 16     footer_string : ''
 17     footer_right_repeat : ''
 ```

I also ran the example with the following JSON file:
```json
  1 {
  2   "rule" : {
  3     "block_comment_001" : {
  4       "disable" : false,
  5       "header_left" : "-",
  6       "header_left_repeat" : "-",
  7       "header_string" : "",
  8       "header_right_repeat" : ""
  9     },
 10     "block_comment_002" : {
 11       "disable" : false,
 12       "comment_left" : "!"
 13     },
 14     "block_comment_003" : {
 15       "disable" : false,
 16       "footer_left" : "-",
 17       "footer_left_repeat" : "-",
 18       "footer_string" : "",
 19       "footer_right_repeat" : ""
 20     }
 21   }
 22 }
```
...and it worked.

So I think changing the example to use `""` instead of `None` would work for both JSON and YAML.

What do you think?

Regards,

--Jeremy

JHertz5, 2023-08-06:
Hi @jeremiah-c-leary. Excellent, that is a perfect solution! Just to be sure, I've run these examples as well and I agree that `""` would work for both JSON and YAML. I'll put together a PR to change the example to save you the trouble.

jeremiah-c-leary, 2023-08-06:
Awesome, thanks.

--Jeremy

JHertz5, 2023-08-06:
PR #986 raised to resolve this issue.

PR #986 (2023-08-06): Issue#981 Corrected None to "" in block comment example configurations.
**Description**
Correction to example configuration in the Configuring Block Comments page of the docs. See Issue #981 for details.

**Screenshots**
See Issue #981 

**Additional context**
None

Resolves #981
closed 2023-08-06 by jeremiah-c-leary