jeremiah-c-leary/vhdl-style-guide #988: `comment_line_ends_group` doesn't work for some keyword alignment rules handling inline comments

JHertz5, 2023-08-06:
**Environment**
v3.16.0

**Describe the bug**
With `comment_line_ends_group` set to `true`, the comment lines do not break alignment groups for the following rules:
- instantiation_029
- entity_020
- component_020
- package_401
- architecture_027
- (maybe others?)

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file `test.yml` with the following contents
```yamlrule:
  instantiation_023:
    disable: true
  component_019:
    disable: true
  instantiation_029:
    comment_line_ends_group: true
  component_020:
    comment_line_ends_group: true
  entity_020:
    comment_line_ends_group: true
  package_401:
    comment_line_ends_group: true
  architecture_027:
    comment_line_ends_group: true
```
Note that this disables rules removing inline comments from components and instantiations, and sets the value of `comment_line_ends_group` to `true` for each of the rules in question.
2. Create a file `test.vhd` with the following contents
```vhdl

package my_package is

  signal wr_en : std_logic; -- Comment 1
  -- Comment
  signal   rd_en    : std_logic; -- Comment 2
  constant c_period : time;      -- Comment 3

end package my_package;

entity test is
  port (
    i_clk   : in    std_logic; -- Comment 1
    i_rst_n : in    std_logic; -- Comment 2
    -- Comment
    i_module_clk : in    std_logic -- Comment 3
  );
end entity test;

architecture rtl of test is

  signal test_data : std_logic_vector(3 downto 0); -- Comment 1
  -- Comment
  signal pulse : std_logic; -- Comment 2
  signal count : integer;   -- Comment 3

  component mux is
    port (
      sel : in    positive range 1 to n - 1;        -- Comment 1
      din : in    std_logic_vector(n - 1 downto 0); -- Comment 2
      -- Comment
      q : out   std_logic -- Comment 3
    );
  end component mux;

begin

  cmp_mux : component mux
    port map (
      sel => sel,          -- Comment 1
      din => din_external, -- Comment 2
      -- Comment
      q => q -- Comment 3
    );

end architecture rtl;
```
I expect this to report no errors.
3. Run `vsg -f test.vhd -c test.yml`
4. Observe that all of the aforementioned rules are reporting errors.
```
================================================================================
File:  .\test.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 522
Total Violations:    8
  Error   :     8
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  package_401               | Error      |          4 | Move 5 columns
  entity_020                | Error      |         13 | Move 4 columns
  entity_020                | Error      |         14 | Move 4 columns
  architecture_027          | Error      |         24 | Move 23 columns
  architecture_027          | Error      |         25 | Move 23 columns
  architecture_027          | Error      |         32 | Move 26 columns
  component_020             | Error      |         32 | Move 26 columns
  instantiation_029         | Error      |         43 | Move 14 columns
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
I expect the VHDL snippet to report no errors when used with the configuration described.

**Screenshots**
None

**Additional context**
None

jeremiah-c-leary, 2023-10-23:
Good Afternoon @JHertz5 ,

It turns out my testing was insufficient to capture this error.  I pushed an update to the `issue-988` branch for this.  When you get a chance could you run it on your end and let me know if it works.

BTW, There were two other rules affected:  `process_034` and `process_035`.

I am exploring whether to set the default value of `comment_line_ends_gorup` to `no`.  Hoping it will have the same behavior as the error case.  This would minimize the affects to other users.  Not sure if this is the right approach yet though.

Thanks,

--Jeremy

JHertz5, 2023-10-24:
Hi @jeremiah-c-leary. I've tested your branch and it looks to be working perfectly! Thank you so much.

That's a tricky one. I can see the argument for either case. If I were in your position, I think I would leave the defaults as they are, but I have no problem with you changing them.

jeremiah-c-leary, 2023-10-24:
Morning @JHertz5 ,

Setting the  default of `comment_line_ends_group` to `no` resulted in more test failures than I was expecting.  I think I will keep the default to `yes` and try to let users know about the change in behavior on the next release.

I will merge this to master.

--Jeremy

PR #1026 (2023-10-24): Issue#988:  Fixing comment_line_ends_group option for alignment rules.
  * archtiecture_027
  * entity_020
  * component_020
  * instantiation_029
  * package_401
  * process_034
  * process_035

Resolves #988
closed 2023-10-24 by jeremiah-c-leary