jeremiah-c-leary/vhdl-style-guide #990: `constant_016` doesn't raise errors when there is a constraint in the type of the constant declaration.

JHertz5, 2023-08-06:
**Environment**
3.16.0

**Describe the bug**
When there is a contraint in the type of a constant declaration, e.g. std_logic_vector **(3 downto 0)**, rule `constant_016` doesn't appear to be activated.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file `test.vhd` with the following contents
```vhdl
architecture rtl of test is

  constant c_data : signed(t_data_range) := (o_data'high => '1',
                                              others => '0');

  constant c_data : t_data := (o_data'high => '1',
                                others => '0');

begin

end architecture rtl;
```
I expect this snippet to raise some errors for `constant_016` as it does not follow the default settings.
2. Run `vsg -f test.vhd`
3. Observe the following output
```
================================================================================
File:  .\test.vhd
================================================================================
Phase 5 of 7... Reporting
Total Rules Checked: 524
Total Violations:    3
  Error   :     3
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  constant_016              | Error      |          6 | Move parenthesis after assignment to the next line.
  constant_016              | Error      |          6 | Add carriage return after open parenthesis.
  constant_016              | Error      |          7 | Move closing parenthesis to the next line.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
4. See that, although the structure of the two constant declarations is the same, errors have been raised only for the second declaration and not the first.

**Expected behavior**
I expect `constant_016` to treat these constant declarations equally.

**Screenshots**
None

**Additional context**
None

jeremiah-c-leary, 2023-10-25:
Good Afternoon @JHertz5 ,

This was a subtle issue to find.  I pushed an update to the `issue-990` branch.  When you get a chance could you run it on your end and let me know how it goes?

Thanks,

--Jeremy

JHertz5, 2023-10-25:
Hi @jeremiah-c-leary. Thanks so much for looking in to this. I have had a try with your branch and it works perfectly.

jeremiah-c-leary, 2023-10-26:
Awesome, I will merge this to master.

--Jeremy

PR #1029 (2023-10-26): Issue#990:  Rule constant_016 ignored if subtype_indication included parenthesis.
Resolves #990
closed 2023-10-26 by jeremiah-c-leary