jeremiah-c-leary/vhdl-style-guide #994: Rule for function call structure

JHertz5, 2023-08-07:
**Is your feature request related to a problem? Please describe.**
I would like to have a rule to enforce the structure of function calls, similar to `procedure_call_003`. As far as I can tell, there are no rules governing function calls as of v3.16.0.

**Describe the solution you'd like**
Input
```vhdl
architecture rtl of test is

begin

  test <= my_function(        arg1            =>         input1,
                   arg2       =>       input2);

  my_procedure(        arg1            =>         input1,
                   arg2       =>       input2);

end architecture rtl;
```

Current output (with default configuration)
```vhdl
architecture rtl of test is

begin

  test <= my_function(arg1 => input1,
                   arg2    => input2);

  my_procedure(arg1 => input1,
               arg2 => input2);

end architecture rtl;
```
Note that the corrections made to the function call on the alignment of the `=>`s and the inputs are due to rules `concurrent_400` (which is intended for record assignments) and `element_association_101`.

Desired output
```vhdl
architecture rtl of test is

begin

  test <= my_function(arg1 => input1,
                      arg2 => input2);

  my_procedure(arg1 => input1,
               arg2 => input2);

end architecture rtl;
```

**Describe alternatives you've considered**
Since rules `concurrent_400` and `element_association_101` do some of the work required, a potentially simpler solution would be to add a rule that aligns the formal parameters in the function call.

**Additional context**
None

jeremiah-c-leary, 2023-11-11:
Good Afternoon @JHertz5 ,

I looks like my array detection was incorrect.  I pushed an update to the `issue-994` branch.  Could you check it out and let me know if it addresses the issue on your side.

Thanks,

--Jeremy

JHertz5, 2023-11-12:
Hi @jeremiah-c-leary. I have tested your branch and it is working perfectly! Thanks for taking the time to work on this.

jeremiah-c-leary, 2023-11-12:
Morning @JHertz5 ,

Awesome, I will merge this into master.

--Jeremy

PR #1044 (2023-11-12): Issue#994:  concurrent_003 was misdetecting arrays.
Resolves #994
closed 2023-11-12 by jeremiah-c-leary