jeremiah-c-leary/vhdl-style-guide #998: rule `variable_011` fails with multiple procedures in same file

t-aras, 2023-08-14:
**Environment**
Version 3.15.0
OS: CentOS

**Describe the bug**
The rule `variable_011` fails with multiple procedures in the same file.
When a variable and parameter, both in different procedures, have the same name (let's say `addr`) but different capitalization, rule `variable_011` takes the capitalization of the variable also for the procedure where `addr` is used as a parameter.

**To Reproduce**
Use example file here:

```vhdl
entity test is
end entity test;

architecture arch of test is

begin

  ps_test0 : process is

    ----------------------------------------------------------------------------
    -- procedure_1

    procedure procedure_1 (
      constant INSN_BASE : in integer;
      constant ADDR      : in integer
    ) is

      variable test_var0 : integer;
      variable test_var1 : integer;

    begin

      test_var0 := INSN_BASE;
      test_var1 := ADDR;

    end procedure procedure_1;

  begin -- ps_test0

  end process ps_test0;

  ps_test1 : process is

    ----------------------------------------------------------------------------
    -- procedure_2

    procedure procedure_2 is

      variable addr      : integer;
      variable insn_base : integer;

    begin

    end procedure procedure_2;

  begin -- ps_test1

  end process ps_test1;

end architecture arch;

```

This triggers following error:

```
vsg -f testfile.vhd  
================================================================================
File:  testfile.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 714
Total Violations:    2
  Error   :     2
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  variable_011              | Error      |         23 | Change "INSN_BASE" to "insn_base"
  variable_011              | Error      |         24 | Change "ADDR" to "addr"
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

**Expected behavior**
I would expect no error at all. The capitalization of the parameters is consistent within its scope (ADDR, INSN_BASE). It seems that vsg is confused by other procedures in the same file that use a variable `addr`.

t-aras, 2023-11-09:
I have found another case where this rule fails:

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package pkg_test is

  -- Helper function
  function function1 (
    constant ADDR : std_logic_vector
  ) return std_logic_vector;

  function function2 (
    constant MY_ADDR : natural
  ) return std_logic_vector;

end package pkg_test;

package body pkg_test is

  function function1 (
    constant ADDR : std_logic_vector
  ) return std_logic_vector is
  begin
    return ADDR;
  end function function1;

  function function2 (
    constant MY_ADDR : natural
  ) return std_logic_vector is
    variable addr : std_logic_vector(31 downto 0);
  begin
    addr := std_logic_vector(to_unsigned(MY_ADDR, addr'length));
    return addr;
  end function function2;

end package body pkg_test;
```

with following error message:

```
================================================================================
File:  ./pkg_test.vhd
================================================================================
Phase 6 of 7... Reporting
Total Rules Checked: 702
Total Violations:    1
  Error   :     1
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  variable_011              | Error      |         32 | Change "ADDR" to "addr"
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```

jeremiah-c-leary, 2023-11-25:
Good Afternoon @t-aras ,

I pushed an update for this issue to the `issue-551` branch.  When you get a chance could you check it on your end and let me know if it resolves the issues on your end?

Just and FYI, there are three other issues related to this one:  #551, #775 and #1023 .  I will close this issue when all four are resolved.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @t-aras ,

Just wanted to ping you on this issue to see if you had time to check it out.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-01-21:
Afternoon @t-aras ,

I just wanted to let you know that I will be merging to this master and closing the issue.  If you find additional problem, please create another issue.

Regards,

--Jeremy

t-aras, 2024-02-04:
Dear @jeremiah-c-leary 

Sorry for the very late reply.
I just tested this fix and this one is also working.
Thank you for this fix too!
Much appreciated!

Best
@t-aras

PR #1052 (2024-01-21): Issue#551:  Fixing scoping of consistent token case rules.
Resolves #551
Resolves #755
Resolves #998 
Resolves #1023 
Resolves #1062