jeremiah-c-leary/vhdl-style-guide #999: No columns-movement for signals inside constant declrations

ungultig1, 2023-08-30:
Hello,

i checked settings for "constant"-adjustments, but could not find any solution regarding this issue:

It moves columns for the record, but not for the constant declaration:
`
type t_my_rec is record
    signal_one       : std_logic;
    signal_onetwo : std_logic;
end record my_rec ;

constant my_rec : t_my_rec :=
(
    signal_one => '0',
    signal_onetwo => '0'
);
`

Expected:
`
type t_my_rec is record
    signal_one       : std_logic;
    signal_onetwo : std_logic;
end record my_rec ;

constant my_rec : t_my_rec :=
(
    signal_one       => '0',
    signal_onetwo => '0'
);
`

Any plans to include this also?

jeremiah-c-leary, 2023-10-29:
Good Morning @ungultig1 ,

I pushed an update for this issue to the `issue-999` branch.

I added rule `constant_400` which will align the `=>`  in constant declarations.

When you get a chance could you check it out and let me know how it goes?

Thanks,

--Jeremy

ungultig1, 2023-11-06:
Good Afternoon @jeremiah-c-leary ,

i tested the column-movement for constant declarations.
It works perfect.

Thank you for fixing.

Greetings,

jeremiah-c-leary, 2023-11-06:
Awesome, I will merge this to master.

--Jeremy

PR #1039 (2023-11-06): Issue#999:  Adding rule constant_400 to align => in constant declarations
Resolves #999
closed 2023-11-06 by jeremiah-c-leary