jeremiah-c-leary/vhdl-style-guide #1022: Function_018 seems to produce invalid VHDL Code

masubious, 2023-10-16:
**Environment**
VSG Version: VHDL Style Guide (VSG) version: 3.17.0+zip.file
Python: 3.10.7 (tags/v3.10.7:6cc6b13, Sep  5 2022, 14:08:36) [MSC v.1933 64 bit (AMD64)]

**Describe the bug**
 Rule Function_018 can produce invalid code when previous functions are ended with "end;" instead of "end function;". Seems to occur if there are 2 functions / procedures in a row ended with just an "end;" and seems to repeat the last token of the following function / procedure. After Beautify of the test_pkg.vhd the procedure proc_a should conclude with "END PROCEDURE proc_a; proc_a;" and thus produce invalid Code.

**To Reproduce**
VSG test_pkg --fix (Default configuration had the same behaviour, reproducing the bug

Test package:
```
PACKAGE test_pkg IS

  FUNCTION func_a (x : boolean; y : boolean) RETURN boolean;

  FUNCTION func_b (x : boolean; y : boolean) RETURN boolean;

  PROCEDURE proc_a (x : IN boolean; SIGNAL y : OUT boolean);

  PROCEDURE proc_b (x : IN boolean; SIGNAL y : OUT boolean);

END PACKAGE test_pkg;

PACKAGE BODY test_pkg IS

  FUNCTION func_a (x : boolean; y : boolean) RETURN boolean IS
  BEGIN

    v_ret := x OR y;
    RETURN v_ret;

  END;

  FUNCTION func_b (x : boolean; y : boolean) RETURN boolean IS
  BEGIN

    v_ret := x AND y;
    RETURN v_ret;

  END;

  PROCEDURE proc_a (x : IN boolean; SIGNAL y : OUT boolean) IS
  BEGIN

    y <= x;

  END PROCEDURE proc_a;

  PROCEDURE proc_b (x : IN boolean; SIGNAL y : OUT boolean) IS
  BEGIN

    y <= NOT x;

  END PROCEDURE proc_b;

END PACKAGE BODY test_pkg;

````

**Expected behavior**
Produce valid VHDL Code after fixing.

**Additional context**
-

jeremiah-c-leary, 2023-10-25:
Good Morning @masubious,

Thank you for finding this issue.  My testing was not as thorough as I had hoped.

I pushed an update for this to the `issue-1022` branch.  When you get a chance could you check it on your end and let me know how it works?

I still need to review other rules that could have the same issue, so I will be pushing updates if needed.

Thanks,

--Jeremy

masubious, 2023-10-30:
Hey @jeremiah-c-leary 
I've checked it based on the example I provided. In Default Configuration Rule Function_018 is set to "add" and correctly Adds the "Function" Keyword to the "End;" Lines. But probably there is a second Rule needed for the Function Name, which doesn't get added anymore ("end function; instead of end function func_a;)

jeremiah-c-leary, 2023-10-31:
Morning @masubious ,

> But probably there is a second Rule needed for the Function Name, which doesn't get added anymore ("end function; instead of end function func_a;)

Could you create another issue for this.  I am using this branch to address issue #950  also.  Then I can merge in this update and will work on the new issue.

Thanks,

--Jeremy

PR #1028 (2023-11-02): Issue#1022:  Fixing rule function_018.
  * Rule could not determine the difference between subprogram_body and subprogram_declaration.

Resolves #1022 
Resolves #950
closed 2023-11-02 by jeremiah-c-leary