jeremiah-c-leary/vhdl-style-guide #1034: Array Multiline Structure Rules do not handle

JHertz5, 2023-10-28:
**Environment**
master branch
```
VHDL Style Guide (VSG) version: 3.17.0.dev19
Git commit SHA: 92503eee
```

**Describe the bug**
When presented with an aggregate assignment in which an array element is referenced by an expression that include parenthesis (e.g. a value of another array), the rule treats that as the start of a sub-array. This results in behaviour that is likely not desired by the user.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file called `test.vhd` with the following contents:
```vhdl
architecture rtl of test is

  constant c_constant : t_array :=
  (
    c_enum_list(ENUM_LITERAL_1) => 1,
    others => 'X'
  );

  signal data_concurrent  : t_array;
  signal data_sequuential : t_array;

begin

  data_concurrent <=
  (
    c_enum_list(ENUM_LITERAL_1) => 1,
    others       => 'X'
  );

  proc_test : process (all) is

    variable v_variable : t_array;

  begin

    v_variable :=
    (
      c_enum_list(ENUM_LITERAL_1) => 1,
      others => 'X'
    );

    data_sequuential <=
    (
      c_enum_list(ENUM_LITERAL_1) => v_array(1),
      others       => 'X'
    );

  end process proc_test;

end architecture rtl;

```
2. Run `vsg -f ./test.vhd` and observe the following output:
```
================================================================================
File:  ./test.vhd
================================================================================
Phase 1 of 7... Reporting
Total Rules Checked: 157
Total Violations:    6
  Error   :     6
  Warning :     0
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  concurrent_012            | Error      |         16 | Add carriage return after open parenthesis.
  concurrent_012            | Error      |         16 | Move closing parenthesis to the next line.
  variable_assignment_008   | Error      |         28 | Add carriage return after open parenthesis.
  variable_assignment_008   | Error      |         28 | Move closing parenthesis to the next line.
  sequential_009            | Error      |         34 | Add carriage return after open parenthesis.
  sequential_009            | Error      |         34 | Move closing parenthesis to the next line.
----------------------------+------------+------------+--------------------------------------
NOTE: Refer to online documentation at https://vhdl-style-guide.readthedocs.io/en/latest/index.html for more information.
```
5. Run `vsg -f ./test.vhd --fix` and observe the following changes to test.vhd:
```vhdl
architecture rtl of test is

  constant c_constant : t_array :=
  (
    c_enum_list(
                 ENUM_LITERAL_1
               ) => 1,
    others => 'X'
  );

  signal data_concurrent  : t_array;
  signal data_sequuential : t_array;

begin

  data_concurrent <=
  (
    c_enum_list(
                 ENUM_LITERAL_1
               ) => 1,
    others       => 'X'
  );

  proc_test : process (all) is

    variable v_variable : t_array;

  begin

    v_variable :=
    (
      c_enum_list(
                   ENUM_LITERAL_1
                 ) => 1,
      others => 'X'
    );

    data_sequuential <=
    (
      c_enum_list(
                   ENUM_LITERAL_1
                 ) => v_array(1),
      others       => 'X'
    );

  end process proc_test;

end architecture rtl;
```

**Expected behavior**
I would expect the expression `c_enum_list(ENUM_LITERAL_1)` to be unaffected by these rules since these parenthesis do not create a sub-array within the array.

jeremiah-c-leary, 2023-11-12:
Good Evening @JHertz5 ,

I pushed an update to the `issue-1034` branch.  When you get a chance could you check it out and let me know if there are any other issues.

Thanks,

--Jeremy

JHertz5, 2023-11-12:
Hi @jeremiah-c-leary. Thanks so much for taking a look at this. I have run your branch and it looks perfect!

jeremiah-c-leary, 2023-11-12:
Morning @JHertz5 ,

Awesome, I will merge this into master.

--Jeremy

PR #1045 (2023-11-12): Issue#1034:  Improving classification of functions and procedures.
Resolves #1034
closed 2023-11-12 by jeremiah-c-leary