jeremiah-c-leary/vhdl-style-guide #1040: Rename signal as they are defined

ungultig1, 2023-11-01:
Goog morning @jeremiah-c-leary,

I wonder, if the following feature can already be activated (and I missed it) or please could be implemented (?):

At the moment the signals-naming is defined by vsg as default.

Is there any possibility, that the signals-name are being formatted by looking at the definitions?

For instance:

```
architecture behaviour of example is
...
...

signal sl_mySignalNaming : std_logic; -- my defined signal-naming

begin

-- vsg-default after format:
sl_mysignalnaming <= '1'

-- Propose:
-- a setting, which allows to format the signals-naming as defined above:
sl_mysignalNaming <= '1' -- this is the wrong case while coding
-- after format, this will be corrected to:
sl_mySignalNaming <= '1' -- as defined

end behaviour;
```

As i prefer the camel-case-notation and vsg of cource couldn't know how to keep the correct naming, vsg could keep a look at the signals-name definition (also considered the definitions in record-, port- signal-namings etc.)

Big thanks to you.

Greetings,

ungultig1, 2023-11-01:
Hello @jeremiah-c-leary ,

yeah, deactivating signal_004-rule made the difference. Great, Thanks.

Unfortunately, it wont format the signal-names, when defined inside a record.
There might be another rule?

Example:

```
architecture behaviour of example is
...
..
type t_myrec is record 
    sl_mySignalNaming : std_logic; -- my defined signal-naming
end record t_myrec;

constant myrec : t_myrec :=
(
    sl_mySignalnaming => '0' --- After Format, wont apply my definition here.
);

signal   rec                           : t_myrec;

begin

-- After Format, wont apply my definition here also:
rec.sl_mySignalnaming <= '1'

end behaviour;
```

maybe I'm missing here something else.

Thank you.

Greetings,

jeremiah-c-leary, 2023-11-01:
Good Afternoon @ungultig1 ,

There are currently no rules to enforce consistent casing in your latest example.  I currently do not have a rule similar to what you are requesting so it will take a bit to implement.  I have to ensure I keep the scoping correct.  I think I have to plan on how to do it.

--Jeremy

ungultig1, 2023-12-18:
> Hello @jeremiah-c-leary ,
> 
> yeah, deactivating signal_004-rule made the difference. Great, Thanks.
> 
> Unfortunately, it wont format the signal-names, when defined inside a record. There might be another rule?
> 
> Example:
> 
> ```
> architecture behaviour of example is
> ...
> ..
> type t_myrec is record 
>     sl_mySignalNaming : std_logic; -- my defined signal-naming
> end record t_myrec;
> 
> constant myrec : t_myrec :=
> (
>     sl_mySignalnaming => '0' --- After Format, wont apply my defined signal-naming here.
> );
> 
> signal   rec                           : t_myrec;
> 
> begin
> 
> -- After Format, wont apply my defined signal-naming here also:
> rec.sl_mySignalnaming <= '1'
> 
> end behaviour;
> ```
> 
> maybe I'm missing here something else.
> 
> Thank you.
> 
> Greetings,

Good morning @jeremiah-c-leary,

were you able to create such rule? 
A rule, which does the name-formatting for used record signals as it is declared inside the record-structure? 

Thank you.
Greetings.

jeremiah-c-leary, 2023-12-18:
Afternoon @ungultig1 ,

At the moment I am having an issue with scoping and ensuring a consistent case can be applied correctly when signals with similar names are used.  I do not have a reasonable solution yet that does not require a significant update to the internal data structure.

I believe it is something that should be implemented, I'm afraid it will take awhile before it finally is implemented.

Regards,

--Jeremy

jeremiah-c-leary, 2023-12-22:
Morning @ungultig1 ,

So I have been thinking about this a little more and it seems there are two parts you asking for:

1. control case and prefix/suffix of elements of a record
2. ensure the same case is applied throughout the code

I believe the first one should not be that hard, and I want to think the second would could be done with my existing data structure.  I just need to think on it a little more.

Regards,

--Jeremy