jeremiah-c-leary/vhdl-style-guide #1047: Case-rule for either 'upper' or 'lower', but not a combination

larshb, 2023-11-14:
When using the `use`-keyword I will sometimes import components (lowercase) and sometimes import constants (uppercase). But I can't have both without disabling `use_clause_502`.

I would like to be able to set enforcement of "either". Eg.
```yaml
rule :
  use_clause_502:
     case: 'either'
```

resulting in

```vhdl
library mylib;
  use mylib.mypackage.C_SOME_CONSTANT; -- OK
  use mylib.mypackage.some_component; -- OK
  use mylib.mypackage.sOmEtHiNg_ElSe; -- violation
```

jeremiah-c-leary, 2023-11-14:
Morning @larshb ,

I added an option `upper_or_lower` that will check for upper or lower cases.

I pushed an update to the `issue-1047` branch, when you get a chance could you check it out on your end and let me know what you think?

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-09:
Morning @larshb ,

Just wanted to ping you on this issue to see if you had time to check it out.

Thanks,

--Jeremy

jeremiah-c-leary, 2023-12-13:
Morning @larshb ,

Awesome, I will get this merged to master.

--Jeremy

PR #1048 (2023-12-13): Issue#1047:  Adding upper_or_lower option to case rules.
Resolves #1047
closed 2023-12-13 by jeremiah-c-leary