jeremiah-c-leary/vhdl-style-guide #1062: False positive with "type_501"

bae-pabr, 2023-12-06:
**Environment**
vsg                     3.18.0

**Describe the bug**
rule "type_501" is triggered, which is I think a false positive.

**To Reproduce**
1. create file with content:
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

package foo_pkg is

    type t_foo_in is record
        start            : std_logic; -- this is line 8
        stop             : std_logic;
    end record;

end package;

architecture rtl of bar is

    type t_state is (IDLE, START, PWM);

begin

end rtl;
```
2. run: vsg -f t501.vhdl -ap

**Expected behavior**
`  type_501                  | Error      |          8 | Change "start" to "START"`
rule type_501 should not trigger for line 8, because this is not an enum.

jeremiah-c-leary, 2023-12-06:
Morning @bae-pabr ,

I recently made some improvements to the consistent case rules which I believe will address this issue.  Could you try using branch `issue-551` and see if that resolves `type_501` in your case?

Thanks,

--Jeremy

bae-pabr, 2023-12-07:
Hi Jeremy,

I tried branch issue-551 and the issue seems to be fixed for me. When i compare both results type 501 no longer triggers for line 8.

Thanks,
Patrick

jeremiah-c-leary, 2023-12-07:
Evening @bae-pabr ,

That is good news.  I am waiting for others to approve the changes on that branch before I merge it to master.

Hopefully it does not take that long.

--Jeremy

jeremiah-c-leary, 2024-01-21:
Afternoon @bae-pabr ,

Just wanted to let you know that I will be merging this to master.

Regards,

--Jeremy

PR #1052 (2024-01-21): Issue#551:  Fixing scoping of consistent token case rules.
Resolves #551
Resolves #755
Resolves #998 
Resolves #1023 
Resolves #1062
closed 2024-01-21 by jeremiah-c-leary