jeremiah-c-leary/vhdl-style-guide #1066: possible typo in process_021 python

SittingDuc, 2023-12-13:
Hello again,

Reviewing the configuration file changes from 3.18.0 to 3.19.0, I find process_021 has a default of "no_blank_line_line", line 64 of vsg/rules/process/rule_021.py

```
     60     def __init__(self):
     61         blank_line.Rule.__init__(self, 'process', '021')
     62         self.solution = 'Remove blank lines above *begin* keyword'
     63         self.phase = 1
     64         self.style = 'no_blank_line_line'
     65         self.configuration.append('style')
```

Given that "line_line" does not appear in any other file in the project, I expect this is a typo?

```
VHDL Style Guide (VSG) version: 3.19.0
Git commit SHA: c3e408e9
```

copied this morning

Cheers!

jeremiah-c-leary, 2023-12-13:
Morning @SittingDuc ,

Good catch.  I pushed an update to the `issue-1066` branch.  When you get a chance could you validate it on your end?

Thanks,

--Jeremy

SittingDuc, 2023-12-13:
Hello. I have eyeballed the commit (but not run it up) and I see no typos in the fix for the typo :)
Thanks

jeremiah-c-leary, 2023-12-14:
Morning @SittingDuc ,

I will merge this to master.

--Jeremy

PR #1067 (2023-12-14): Issue#1066:  Fixing default style of rule process_021
Resolves #1066
closed 2023-12-14 by jeremiah-c-leary