jeremiah-c-leary/vhdl-style-guide #1074: Is there a rule similar to procedure_calL_003 for sequential procedure calls?

JHertz5, 2023-12-20:
**What is your question?**
I would like to enforce the structure of procedure calls within processes, similar to how procedure_calL_003 behaves for concurrent procedures. I haven't been able to find such a rule, does this exist already?

jeremiah-c-leary, 2023-12-20:
Morning @JHertz5 ,

The rule `procedure_call_003` should have also applied to sequential procedure calls.  I checked my tests and for some reason the tests were not validating it applied to sequential procedure calls.  I updated the tests and rule `procedure_call_003` to handle them.

I pushed an update to the `issue-1074` branch.  When you get a chance could  you check it out and let me know if there are any issues?

Thanks,

--Jeremy

JHertz5, 2023-12-20:
Hi @jeremiah-c-leary. That was fast, thanks very much! I've tested your branch with the test case that I described in #1073, and it is working perfectly!

JHertz5, 2023-12-20:
With this change, perhaps the references to "concurrent procedure calls" in the documentation for this rule should be removed, to avoid implying that sequential procedure calls are still excluded. I would be happy to raise a separate issue and make those changes myself.

jeremiah-c-leary, 2023-12-20:
Afternoon @JHertz5 ,

I think it would  be best to update the documentation as part of this issue.  I have pushed some updates, could you check and see if I missed anything?

Thanks,

--Jeremy

JHertz5, 2023-12-21:
Hi @jeremiah-c-leary. That makes sense 👍 It looks good to me, thanks very much!

jeremiah-c-leary, 2023-12-21:
Morning @JHertz5 ,

Awesome, I will get this merged to master.

--Jeremy

PR #1075 (2023-12-21): Issue#1074:  procedure_call_003 now includes sequential procedure calls.
Resolves #1074
closed 2023-12-21 by jeremiah-c-leary