jeremiah-c-leary/vhdl-style-guide #1076: Add an "ignore_single_line" option to prcoedure_call_003

JHertz5, 2023-12-23:
**Is your feature request related to a problem? Please describe.**
Many rules, such as [Array Multiline Structure Rules](https://vhdl-style-guide.readthedocs.io/en/3.20.0/configuring_array_multiline_structure_rules.html#configuring-array-multiline-structure-rules) have an `ignore_single_line` option. I would like to have a similar option for the `prcoedure_call_003` rule, so that I can enforce structure on multi-line procedure calls while leaving neat one-liners alone.

**Describe the solution you'd like**
Input:
```vhdl
entity test is
end entity test;

architecture rtl of test is

begin

  lil_proc(actual_1);

  big_proc(
           formal_1 => actual_1,
           formal_2 => actual_2
         );

end architecture rtl;
```
Config
```yaml
rule:
  procedure_call_003:
    first_open_paren: 'remove_new_line'
    last_close_paren: 'add_new_line'
    association_element: 'add_new_line'
    association_list_comma: 'ignore'
    ignore_single_line: 'yes'
```
Current output (ignoring the fact that `ignore_single_line` would break the current config):
```vhdl
entity test is
end entity test;

architecture rtl of test is

begin

  lil_proc(actual_1
         );

  big_proc(
           formal_1 => actual_1,
           formal_2 => actual_2
         );

end architecture rtl;
```
Desired output:
```vhdl
entity test is
end entity test;

architecture rtl of test is

begin

  lil_proc(actual_1);

  big_proc(
           formal_1 => actual_1,
           formal_2 => actual_2
         );

end architecture rtl;
```

JHertz5, 2023-12-23:
I have had a go at actioning this issue and raised PR #1077. Any feedback is welcome.

jeremiah-c-leary, 2024-01-01:
Good afternoon @JHertz5 ,

I reviewed the PR and it looks good.

Thank you for updating the documentation and the tests.

I will merge this to master.

--Jeremy

PR #1077 (2024-01-01): Issue#1076: procedure_call_003 modified to accept ignore_single_line option.
Resolves #1076.

I have set the default value of the new option to `no` to avoid modifying the tool's default behaviour from what exists currently.
closed 2024-01-01 by jeremiah-c-leary