jeremiah-c-leary/vhdl-style-guide #1079: `procedure_410` doesn't act on `file` parameters

JHertz5, 2023-12-23:
**Environment**
VHDL Style Guide (VSG) version: 3.20.0
Git commit SHA: 22beaa35

**Describe the bug**
`procedure_410` aligns the colons of the parameter in procedure declarations, but it fails to act on `file` parameters.

**To Reproduce**
Steps to reproduce the behavior:
1. Create test.vhd
```vhdl
use std.textio.all;

entity test is
end entity test;

architecture rtl of test is

  procedure my_proc (
    constant a      : in integer;
    signal b     : out integer;
    file c         : text;
    variable d   : integer;
    e             : integer
  );

begin

end architecture rtl;
```
2. Run `vsg -f ./test.vhd --fix`
3. Observe that test.vhd is now
```vhdl
use std.textio.all;

entity test is
end entity test;

architecture rtl of test is

  procedure my_proc (
    constant a : in integer;
    signal b   : out integer;
    file c         : text;
    variable d : integer;
    e          : integer
  );

begin

end architecture rtl;
```
4. Note that the colon of the file parameter has not been aligned.

**Expected behavior**
I expect to see
```vhdl
use std.textio.all;

entity test is
end entity test;

architecture rtl of test is

  procedure my_proc (
    constant a : in integer;
    signal b   : out integer;
    file c     : text;
    variable d : integer;
    e          : integer
  );

begin

end architecture rtl;
```

JHertz5, 2023-12-23:
I have had a go at solving this and have raised PR #1080

jeremiah-c-leary, 2024-01-01:
Afternoon @JHertz5 ,

The pull request looks good.

Thanks for updating the test case.

I will get this merged to master.

Regards,

--Jeremy

PR #1080 (2024-01-01): Issue#1079 Updated procedure_410 to act on colons in file interface declarations.
Resolves #1079
closed 2024-01-01 by jeremiah-c-leary