jeremiah-c-leary/vhdl-style-guide #1084: Error in the docstring for `iteration_scheme_501`

JHertz5, 2023-12-28:
**Environment**
State installed version of this project and your OS information.
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0
Git commit SHA: 22beaa35
```

**Describe the bug**
When running `vsg_rule_doc_gen`, the "Iteration Scheme Rules" page's entry for `iteration_scheme_501` changes the description from "This rule checks the **for** keyword has proper case." (correct) to "This rule checks the **while** keyword has proper case." (incorrect). This error originates from the docstring in vsg/rules/iteration_scheme/rule_501.py.

JHertz5, 2023-12-29:
I have had a go at solving this in PR #1085. Any feedback is welcome.

jeremiah-c-leary, 2024-01-02:
Evening @JHertz5 ,

I just added a test for the `iteration_scheme` documentation, othewise it looks good.

I will merge this to master.

Happy New Year,

--Jeremy

PR #1085 (2024-01-02): Issue#1084: Updated docstring for `iteration_scheme_501`.
Resolves #1084.
I also updated the iteration_scheme_rules.rst page with a `-` that was present in the autogenerated page but missing from the page in source control.
closed 2024-01-02 by jeremiah-c-leary