jeremiah-c-leary/vhdl-style-guide #1086: Errors in the examples on the "Configuring Multiline Indent Rules" page.

JHertz5, 2023-12-28:
**Environment**
State installed version of this project and your OS information.
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev2
Git commit SHA: cba5db27
```

**Describe the bug**
The [Configuring Multiline Indent Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html#configuring-multiline-indent-rules) page displays examples to show the behaviour of each combination of the `align_left` and `align_paren` options. However, some of these examples are incorrect. Namely:
- The example for "`align_left` set to no and `align_paren` set to no" is actually showing the example for "`align_left` set to no and `align_paren` set to yes" (and therefore the two examples are the same).
- The example for "`align_left` set to yes and `align_paren` set to yes" does not align against the parenthesis.

As a side note, it is a bit confusing IMO to show the configuration with the rule `constant_012` and then to show examples that are not affected by `constant_012`, but rather by `concurrent_003`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html#configuring-multiline-indent-rules
2. Copy the example snippet into a file called test.vhd with a surrounding architecture and entity, producing the following contents:
```vhdl
entity test is
end entity test;

architecture rtl of test is

begin

  wr_en <= resize(unsigned(I_FOO) +
              unsigned(I_BAR), q_foo'length);

  wr_en <=
      resize(unsigned(I_FOO) +
                      unsigned(I_BAR), q_foo'length);

end architecture rtl;
```
3. Create a file called test.yml with the following contents:
```yaml
rule :
  concurrent_003:
     align_left : 'yes'
     align_paren : 'yes'
  concurrent_011:
    disable: true
```
4. Run  `./bin/vsg -c ./test.yml -f ./test.vhd --fix` and compare test.vhd to the example on the webpage.
5. Try this with each configuration

**Expected behavior**
The documentation should show the following for each configuration:
- align_left: yes, align_paren: no (this does match the current documentation ✔️)
```vhdl
  wr_en <= resize(unsigned(I_FOO) +
      unsigned(I_BAR), q_foo'length);

  wr_en <=
    resize(unsigned(I_FOO) +
      unsigned(I_BAR), q_foo'length);
```
- align_left: no, align_paren: no (this **does not** match the current documentation ❌)
```vhdl
  wr_en <= resize(unsigned(I_FOO) +
             unsigned(I_BAR), q_foo'length);

  wr_en <=
           resize(unsigned(I_FOO) +
             unsigned(I_BAR), q_foo'length);
```
- align_left: yes, align_paren: yes (this **does not** match the current documentation ❌)
```vhdl
  wr_en <= resize(unsigned(I_FOO) +
                  unsigned(I_BAR), q_foo'length);

  wr_en <=
    resize(unsigned(I_FOO) +
           unsigned(I_BAR), q_foo'length);
```
- align_left: no, align_paren: yes (this does match the current documentation ✔️)
```vhdl
  wr_en <= resize(unsigned(I_FOO) +
                  unsigned(I_BAR), q_foo'length);

  wr_en <=
           resize(unsigned(I_FOO) +
                  unsigned(I_BAR), q_foo'length);
```

JHertz5, 2023-12-29:
I have had a go at solving this in PR #1087. Any feedback is welcome.

jeremiah-c-leary, 2024-01-02:
Happy New Year @JHertz5 ,

Thanks for finding the issues in my documentation.  It really helps to have someone else review it.

I agree with the changes and will merge it to master.

Regards,

--Jeremy

closed 2024-01-02 by jeremiah-c-leary

JHertz5, 2024-01-02:
> Happy New Year @JHertz5 ,

And to you too @jeremiah-c-leary!

> Thanks for finding the issues in my documentation. It really helps to have someone else review it.

No problem at all, I know how hard it is to catch these kinds of things yourself. We're only human after all! I do really appreciate that there is such thorough documentation for this project, so I'm happy to contribute. 

> I agree with the changes and will merge it to master.

Excellent, thanks very much!

PR #1087 (2024-01-02): Issue#1086: Corrected VHDL examples in "Configuring Multiline Indent Rules"
Resolves #1086