jeremiah-c-leary/vhdl-style-guide #1088: `process_400` doesn't align simple assignment with conditional assignment

JHertz5, 2024-01-03:
**Environment**
```
$ vsg -v
VHDL Style Guide (VSG) version: 3.20.0+zip.file
Git commit SHA: Unknown.  Installed via zip file.
```

**Describe the bug**
In a process that has a simple assignment and a conditonal assignment, `process_400` should align the assignment operators of the assignments. This does not seem to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file test.vhd with the following contents
```vhdl
entity test is
end entity test;

architecture rtl of test is

begin

  proc_label : process (i_clk) is
  begin

    if rising_edge(i_clk) then
      sig        <= i_sig;
      longer_sig <= c_constant when i_select = '1' else
                    i_sig when i_select = '0' else
                    (others => (others => 'X'));
    end if;

  end process proc_label;

end architecture rtl;
```
2. Run `vsg -f test.vhd`
3. Observe that process_400 reports an error on the `sig` assignment, even though it is aligned with the `longer_sig` assignment.

**Expected behavior**
This `process_400` should align the two assignents in this case.

jeremiah-c-leary, 2024-01-07:
Morning @JHertz5 ,

I added the conditional signal assignment to the rule and it appears to be working correctly.

I pushed an update to the `issue-1088` branch.  When you get a chance could you check it out on your end and let me know if it fixes the issue for you.

Thanks,

--Jeremy

JHertz5, 2024-01-07:
HI @jeremiah-c-leary. Thank you so much! That branch is working perfectly for me. I'm happy for the issue to be closed ocne that is merged. Thanks again!

jeremiah-c-leary, 2024-01-07:
Morning @JHertz5 ,

Awesome, I will merge it to master.

--Jeremy

closed 2024-01-07 by jeremiah-c-leary