jeremiah-c-leary/vhdl-style-guide #1095: How to change default indentation from 2 spaces to 3 spaces ?

MJoergen, 2024-01-10:
The default indent_size seems to be 2. I would like to change the desired indent_size to 3.

I've tried using this in the configuration file, but it doesn't work.

```
rule:
  global:
    indent_style : 'spaces'
    indent_size  : 3
```

The expected behavior is that a source file that currently has an indentation of 2 spaces will be changed to having 3 spaces at each indentation. However, the observed behavior is that the source file is unchanged, i.e. still has only 2 spaces of indentation.

My complete configuration file is here (file extension changed from .yml to .txt, apparently required by GitHub)
[vsg.txt](https://github.com/jeremiah-c-leary/vhdl-style-guide/files/13885441/vsg.txt)

JHertz5, 2024-01-10:
Hi @MJoergen. I think I know why your configuration file isn't working. The properties are not called `indent_style` and `indent_size`, they are called `indentStyle` and `indentSize`. You will see that if you run `vsg -rc <rule_name>`, the output should contain 
```
...
      "indentStyle": "spaces",
      "indentSize": 2,
...
```
If you try the following config, you should have better results:
```yml
rule:
  global:
    indentStyle: 'spaces'
    indentSize: 3
```
This mistake is easy to make, given that all, or at least the vast majority, of the properties of rules are snake_case rather than camelCase. Perhaps it is worth raising an issue to correct this inconsistency.

MJoergen, 2024-01-10:
Thank you @JHertz5 .

closed 2024-01-10 by MJoergen