jeremiah-c-leary/vhdl-style-guide #1097: Junit Report says Failure when successfull

albydnc, 2024-01-15:
**Environment**
Fedora 39

**Describe the bug**
The Junit output is reporting failure, while the test is successful.

**To Reproduce**
run `vsg -j ./report.xml test.vhdl`
```vhdl
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;

entity test is
  port (
    clk_i   : in    std_logic;
    reset_i : in    std_logic
  );
end entity test;

architecture rtl of test is

begin

end architecture rtl;
```
**Expected behavior**
I expext the failures  tag in the XML to be 0 and the classname of the testcase to not be failure.

**Screenshots**
CLI output:
```
$ vsg -c vhdl_style.yaml -j report.xml ./test.vhdl 
================================================================================
File:  ./test.vhdl
================================================================================
Phase 7 of 7... Reporting
Total Rules Checked: 732
Total Violations:    0
  Error   :     0
  Warning :     0
```

Junit output:
```xml
<?xml version="1.0" ?>
<testsuite errors="0" hostname="blabla" failures="1" timestamp="2024-01-15T06:02:29" tests="1" time="0" name="vhdl-style-guide">
  <properties>
  </properties>
  <testcase name="./test.vhdl" time="0" classname="failure">
  </testcase>
  <system-out>
  </system-out>
  <system-err>
  </system-err>
</testsuite>

```

jeremiah-c-leary, 2024-01-20:
Afternoon @albydnc ,

I was able to duplicate the issue and have a fix in mind.  I was reading up `classname` and it is supposed to be related to the test itself.  Since VSG is just using the JUnit XML to report errors, I think I can drop `classname`.

So I will plan to fix the failure number and remove `classname`.

Let me know if you feel we should keep `classname`.

Regards,

--Jeremy

jeremiah-c-leary, 2024-01-21:
Good Afternoon @albydnc ,

I pushed an update to the `issue_0197` branch for this issue.  When you get a chance, please check it out and let me know if I can merge it to master.

Thanks,

--Jeremy

jeremiah-c-leary, 2024-02-04:
Morning @albydnc ,

Just a ping to see if you had a chance to check out the update for this issue.

Thanks,

--Jeremy

albydnc, 2024-02-06:
Thank you for the fix! It looks good now :)

-A

closed 2024-02-06 by albydnc

jeremiah-c-leary, 2024-02-07:
Awesome,

I will get this merged to master.

--Jeremy