jeremiah-c-leary/vhdl-style-guide #1101: The solution message for `type_008` is misleading

JHertz5, 2024-01-17:
**Environment**
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev8
Git commit SHA: 23b63ebf
```

**Describe the bug**
Rule [type_008](https://vhdl-style-guide.readthedocs.io/en/latest/type_rules.html#type-008) checks the closing parenthesis of multiline enumerated types is on its own line. The solution message for this rule should say something to suggest that, but instead it says "Move enumerated type to the next line". Moving the enumerated type to the next line will not solve the problem. I suspect that this is a copy/paste error from `type_009`.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file test.vhd with that contains the following snippet within an architecture:
```vhdl
    type state_machine is (
      idle,
      write,
      read,
      done);
```
2. Run `vsg -f test.vhd`
3. Observe the output
```
----------------------------+------------+------------+--------------------------------------
  Rule                      |  severity  |  line(s)   | Solution
----------------------------+------------+------------+--------------------------------------
  type_008                  | Error      |         16 | Move enumerated type to the next line.
----------------------------+------------+------------+--------------------------------------
```

**Expected behavior**
I expect a more accurate error message.

JHertz5, 2024-01-17:
I have created PR #1102 to resolve this issue.

jeremiah-c-leary, 2024-01-20:
Afternoon @JHertz5 ,

Thanks for finding that discprepency.

It was probably a result of a copy/paste as I typically do not validate the solution in my tests.

I will merge this to master.

--Jeremy

PR #1102 (2024-01-20): Issue#1101 Updated the solution message for rule type_008.
Resolves #1101.
closed 2024-01-20 by jeremiah-c-leary