jeremiah-c-leary/vhdl-style-guide #1103: Add rules for function and parameter to have the closing parenthesis on its own line

JHertz5, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
[procedure_006](https://vhdl-style-guide.readthedocs.io/en/latest/procedure_rules.html#procedure-006) and [function_300](https://vhdl-style-guide.readthedocs.io/en/latest/function_rules.html#function-300) can both enforce indentation of the closing parenthesis _if it is on its own line_. However, I can't find any rule to enforce that closing parenthesis being on its own line.

**Describe the solution you'd like**
I would like to be able to enforce the closing parenthesis of a function/procedure to be on its own line. For example, I'd like the code:
```vhdl
package my_package is

  function my_function (
    i_arg1 : integer;
    i_arg2 : boolean) return integer;

  procedure my_procedure (
    i_arg1 : integer;
    i_arg2 : boolean);

end package my_package;

package body my_package is

  function my_function (
    i_arg1 : integer;
    i_arg2 : boolean) return integer is
  begin

    return i_arg1;

  end function;

  procedure my_procedure (
    i_arg1 : integer;
    i_arg2 : string) is
  begin

    report i_arg2;

  end procedure;

end package body my_package;
```
to be formatted to
```vhdl
package my_package is

  function my_function (
    i_arg1 : integer;
    i_arg2 : boolean
  ) return integer;

  procedure my_procedure (
    i_arg1 : integer;
    i_arg2 : boolean
  );

end package my_package;

package body my_package is

  function my_function (
    i_arg1 : integer;
    i_arg2 : boolean
  ) return integer is
  begin

    return i_arg1;

  end function;

  procedure my_procedure (
    i_arg1 : integer;
    i_arg2 : string
  ) is
  begin

    report i_arg2;

  end procedure;

end package body my_package;
```

JHertz5, 2024-01-17:
I will try to raise a PR for this. At the moment I am planning a "simple solution" of a rule each for functions and procedures that enforces the new line. I suppose it would be disabled by default to maintain the current default behaviour.

However, this does not allow enforcement of the inverse, i.e. no newline before the closing parenthesis. In order to have that configurability, I think it might be better to just roll the structure rules for functions and procedure together into a [procedure_call_003](https://vhdl-style-guide.readthedocs.io/en/3.20.0/procedure_call_rules.html#procedure-call-003) style structure rule. I would probably need some guidance before trying that, so unless I hear otherwise, I'll just stick with the simple solution.

JHertz5, 2024-01-20:
I have raised PR #1108 for this issue. It is not yet ready for review, as I am still working on the unit tests.

JHertz5, 2024-01-21:
In addition to the closing parenthesis, the new rules also enforce the newlines around the opening parenthesis, interface list semcolons and interface elements. There are not indentation rules for all of these yet, but that PR is already quite large (and much larger than I originally intended), so I think that the missing indentation features should be handled as a separate issue/when someone requests it.

jeremiah-c-leary, 2024-01-24:
Morning @JHertz5 ,

> so I think that the missing indentation rules should be implemented as a separate issue/when someone requests it.

I agree.  The current indent rules will cover the most likely structural styles.

The pull request looks really good.  I appreciate the documentation and the tests you added to cover the new rules.

I just had several minor comments in the PR.  When we address those I will merge this to master.

How did you find the process of adding these rules?  Anything I can do to make it easier or less confusing?

Regards,

--Jeremy

JHertz5, 2024-01-25:
Hi @jeremiah-c-leary

> The comments show up in the conversion tab of the PR. If you scroll towards the bottom you will see the three comments I made.

I can see in the screenshot that you posted that the comment has the "Pending" label. This means that the comments have been added as part of a review which has not been submitted yet, and no one other than yourself can see them until you submit the review.
This is what I see on the PR currently:

---
[image]
---
[image]
---

> Thank you for the feedback on the contributing experience, it is obviously a blind spot for me.

No problem at all!

jeremiah-c-leary, 2024-01-26:
Evening @JHertz5 ,

> I can see in the screenshot that you posted that the comment has the "Pending" label. This means that the comments have been added as part of a review which has not been submitted yet

I just submitted the review, so I hope you can see the comments now.

--Jeremy

JHertz5, 2024-01-26:
Thanks very much @jeremiah-c-leary! I can see those comments now. I have actioned/responded to each and requested your re-review.

PR #1108 (2024-01-27): Issue#1103: Adding structure rules for subprogram specifications.
Resolves #1103.
These additions involve new rules, function_019 and procedure_013. They make the rules function_009 and procedure_011 redundant.
~~This PR is currently a draft, because the unit tests are still being worked on.~~ Ready for review.
closed 2024-01-27 by jeremiah-c-leary