jeremiah-c-leary/vhdl-style-guide #1104: Typo - "superceeced"

JHertz5, 2024-01-19:
**Environment**
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev8
Git commit SHA: 23b63ebf
```

**Describe the bug**
There are a few cases where the word "superceeced" is used in docstrings for deprecated rules. This is isn't a word and appears to be a misspelling of "superseded".
This is present in the following files:
- docs/case_generate_alternative_rules.rst
- docs/case_rules.rst
- docs/if_rules.rst
- vsg/rules/case/rule_021.py
- vsg/rules/case_generate_alternative/rule_300.py
- vsg/rules/if_statement/rule_032.py
- vsg/rules/if_statement/rule_033.py

JHertz5, 2024-01-19:
I have raised PR #1105 to fix these.

jeremiah-c-leary, 2024-01-20:
Afternoon @JHertz5 ,

> Apologies if it's annoying for me to raise tiny issues like this 😅

Grammar and spelling issues are always appreciated.  I try my best, but stuff will always slip through.

--Jeremy

PR #1105 (2024-01-20): Issue#1104 Corrected "superceeced" typo.
Resolves #1104.
closed 2024-01-20 by jeremiah-c-leary