jeremiah-c-leary/vhdl-style-guide #1106: Default configuration of rule `procedure_call_003` does not match that described in the documentation.

JHertz5, 2024-01-20:
**Environment**
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev12
Git commit SHA: 28d2d5a3
```

**Describe the bug**
The actual default values of rule `procedure_call_003` do not match those described in the documentation.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `vsg -rc procedure_call_003`
2. See output
```
$ ./bin/vsg -rc procedure_call_003
{
  "rule": {
    "procedure_call_003": {
      "indentStyle": "spaces",
      "indentSize": 2,
      "phase": 1,
      "disable": false,
      "fixable": true,
      "severity": "Error",
      "first_open_paren": "ignore",
      "last_close_paren": "ignore",
      "association_list_comma": "ignore",
      "association_element": "ignore",
      "ignore_single_line": "no"
    }
  }
}
```
3. Note that the default configuration sets all options to ignore, other than `ignore_single_line` which is set to `no`.
4. See that the page https://vhdl-style-guide.readthedocs.io/en/3.20.0/configuring_procedure_call_statement_rules.html#configuring-procedure-call-statement-rules describes the default value of all of these rules as "remove_new_line".

**Expected behavior**
I would expect the page to reflect the actual default configuration.

JHertz5, 2024-01-20:
I am planning to generate a PR to resolve this issue.

JHertz5, 2024-01-20:
I have opened PR #1109 to resolve this issue.

jeremiah-c-leary, 2024-01-21:
Afternoon @JHertz5 ,

Thanks for the documentation update.  I will get this merged to master.

--Jeremy

PR #1109 (2024-01-21): Issue#1106 Corrections to procedure call docs.
Resolves #1106.
closed 2024-01-21 by jeremiah-c-leary