jeremiah-c-leary/vhdl-style-guide #1107: Support placement of block closing statement relative to openeing statement vertical alignment

alonbl, 2024-01-20:
**Is your feature request related to a problem? Please describe.**

Per discussion https://github.com/jeremiah-c-leary/vhdl-style-guide/discussions/1013, there is no method to control the vertical alignment of a block to be closed at its opening.

**Describe the solution you'd like**

I would like to have a consistent closing of block that is vertical aligned with the label, for example:

```
    process1 : process (
            x1,
            x2,
            x3,
            x4
    ) is
    begin
```
**Describe alternatives you've considered**

I have tried most of the alignment settings while could not find a method to do so. The closest method I found and what we currently use is:

```
    process1 : process (
            x1,
            x2,
            x3,
            x4
                                 ) is
    begin
```

I would love to format the code with well defined blocks in which each block is closed at the column that open it.

Thanks,

jeremiah-c-leary, 2024-01-23:
Good Evening @alonbl ,

I pushed an update to the `issue-1116` branch.  I added the option `align_to` to the rule `process_028`.  If you set this to `current_indent` then you should get the formatting you want.

When you get a chance could you check it out on your end and let me know if it is working for you?

Thanks,

--Jeremy

alonbl, 2024-01-24:
Works perfectly! thank you so much.

Thanks to you my vhdl developers are producing code that I can actually read!

jeremiah-c-leary, 2024-01-24:
> Works perfectly! thank you so much.

Awesome, I will get this merged to master.  I plan to do a release this weekend.

> Thanks to you my vhdl developers are producing code that I can actually read!

Thanks for the kind words.   It's nice to hear the effort is providing value to others.

--Jeremy