jeremiah-c-leary/vhdl-style-guide #1110: Add rules to enforce prefixes and suffixes on for generate/loop parameters

JHertz5, 2024-01-21:
**Describe the solution you'd like**
I would like to be able to enforce a prefix and/or suffix on parameters for generates and loops, e.g.
```vhdl
for index in t_range loop
```
becomes
```vhdl
for lv_index in t_range loop
```

JHertz5, 2024-01-21:
I am working on a PR for this, it should be raised today.

JHertz5, 2024-01-21:
I have raised PR #1111 for this issue.

jeremiah-c-leary, 2024-01-21:
Afternoon @JHertz5 ,

I reviewed the PR and it looks good.  I was initially thinking there could be a single rule for the generate and loop statements since they both share the `parameter_specification` production.  However, then you would not be able to set generate and loops to different prefixes and suffixes.

Overall the PR looks great.
1)  documentation was created/updated
2)  Tests were added
3)  Rules added
4)  Numbering of the rules are consistent with the current numbering method.

You even fixed some of my copy/paste documentation errors.

Excellent job!!!

I will get this merged to master.

Regards,

--Jeremy

PR #1111 (2024-01-21): Issue#1110 Added rules to enforce prefixes and suffixes on generate/loop parameter indentifiers
Resolves #1110
closed 2024-01-21 by jeremiah-c-leary