jeremiah-c-leary/vhdl-style-guide #1112: The list of disabled rules in "Configuring Disabled Rules" is incomplete

JHertz5, 2024-01-21:
**Environment**
```
$ ./bin//vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev8
Git commit SHA: 23b63ebf
```

**Describe the bug**
There are many disabled rules missing from the list at the bottom of [Configuring Disabled Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_disabled_rules.html).

**To Reproduce**
Steps to reproduce the behavior:
1. [Configuring Disabled Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_disabled_rules.html).
2. Observe that disabled rules [type_100](https://vhdl-style-guide.readthedocs.io/en/latest/type_rules.html#type-100), [type_200](https://vhdl-style-guide.readthedocs.io/en/latest/type_rules.html#type-200), and [type_600](https://vhdl-style-guide.readthedocs.io/en/latest/type_rules.html#type-600) are all missing from the list. These are among many other missing rules.

**Expected behavior**
The page should contain a complete list of the disabled rules. A page like this is very difficult to maintain when not automatically generated or tested. Even if the page is corrected now, errors will creep in again unless a test is put in place.

JHertz5, 2024-01-21:
I have raised PR #1113 to correct the page's rule list and to add a unit test to verify the list of rules in the future.

jeremiah-c-leary, 2024-01-21:
Afternoon @JHertz5 ,

> A page like this is very difficult to maintain when not automatically generated or tested. Even if the page is corrected now, errors will creep in again unless a test is put in place.

Completely agree.  Thanks for added a test to enforce the disabled rule list.

I will get this merged to master.

--Jeremy

PR #1113 (2024-01-21): Issue#1112 Corrected list of disabled rules and added unit test to verify list.
Resolves #1112.
closed 2024-01-21 by jeremiah-c-leary