jeremiah-c-leary/vhdl-style-guide #1117: Reducing duplication of rule identifiers in the codebase

JHertz5, 2024-01-24:
**What is your question?**
I have noticed that when a rule is constructed, there is a bit of duplication occurring in the `__init__()`. Take, for example the code for case_001:

```python
from vsg.rules import token_indent

from vsg import token

lTokens = []
lTokens.append(token.case_statement.case_keyword)
lTokens.append(token.case_statement.end_keyword)
lTokens.append(token.case_statement_alternative.when_keyword)

class rule_001(token_indent):

    def __init__(self):
        token_indent.__init__(self, 'case', '001', lTokens)
```
The rule identifier, '001' is in the name of the class, and is hard-coded into the __init__ method. We could use the special variables that are built in to Python classes to extract the '001' string from the name of the class and remove the duplication, and thereby reduce the work, the risk of mistakes, etc that comes with duplication.

JHertz5, 2024-01-24:
I have created a draft PR, #1118 to demonstrate the change. I will update the PR when I have time, likely this weekend, to remove the now-redundant `identifier` input to the `Rule.__init__()` method.

jeremiah-c-leary, 2024-01-25:
Morning @JHertz5 ,

I really like this idea.  I can't count the number of times I have not updated either `name` or `identifier`, which is why there is always a test to check them.  However, ensuring they are correct by using code is a much better solution.

For the `name` attribute, we could check for exceptions and replace.  For example, the `if` rules are in the `if_statement` directory.  

```python
dir_name = get_current_directory_name()
if dir_name == 'if_statement':
    rule_name = 'if'
else:
    rule_name = dirname
```

That could be a stop gap until a real refactoring of rule names could take place.

Or maybe the directory names can change to match the rule name?  For some reason I named the rule directory `if_statement` instead of `if`.  I would like to think there was some conflict with python since `if` is a python keyword. but then again maybe not.

--Jeremy

JHertz5, 2024-01-25:
Hi @jeremiah-c-leary.

Amazing, I'm glad that you like the suggestion! I'll be glad if I can make your life a tiny bit easier.

> For the name attribute, we could check for exceptions and replace. For example, the if rules are in the if_statement directory.

That is a great idea. I'll include this in the PR, and renaming of rules or directories can come later. You might find it preferable to have infrastructure for exceptions; maybe there are cases where you'd like to have a more detailed directory name while keeping the rule name short.

JHertz5, 2024-01-27:
Hi @jeremiah-c-leary, thanks again for your advice. I have marked PR #1118 as "ready for review". There are a lot of files touched, which means that there is a high likelihood of causing conflicts on other branches, but they should be easy to resolve. Apart from the changes in vsg/rule.py and in the tests directory, the changes are just removing the `name` and `identifier` arguments from init functions everywhere.

jeremiah-c-leary, 2024-02-04:
Afternoon @JHertz5 ,

I had to change the `match` to an `if` statement in the `get_rule_name` function to support pre 3.10 python.  Otherwise it looks really good. 
 It sure cleans up with the code by reducing the number of arguments passed to the base rule class.

There were almost a 1000 files changed, so I appreciate the effort you put in to make development easier.  I will have to keep my eye open for any other similar improvements that can be made.

I will push this to master.

Thanks again,

--Jeremy

PR #1118 (2024-02-04): Issue#1117 Modifed initialisation of the `Rule` class's `name` and `identifier` attributes to reduce duplication.
Resolves #1117.
The interesting changes are in vsg/rule.py. In a couple of cases, In a couple of tests and in the test local_rule, I have had to force the values of `name`, `identifier` and `unique_id` since they can't be extracted from the module name or class name in these cases.
closed 2024-02-04 by jeremiah-c-leary