jeremiah-c-leary/vhdl-style-guide #1120: Running the unit tests leaves behind a file in the root of the repository

JHertz5, 2024-01-26:
**Environment**
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.20.0.dev16
Git commit SHA: 4ff8b6a0
```

**Describe the bug**
After running the unit tests, a test file called `actual.json` is left behind. I have narrowed this down to the test `vsg.tests.tool_integration.quality_report.test_main`.

**To Reproduce**
Steps to reproduce the behavior:
1. Run `python -m unittest vsg.tests.tool_integration.quality_report.test_main`
2. Observe that a new file, `actual.json` is left behind in the root of the repo

**Expected behavior**
I would expect the test to clean up any temporary files upon completion.

JHertz5, 2024-01-26:
I have raised PR #1121 to resolve this.

jeremiah-c-leary, 2024-01-27:
Morning @JHertz5 ,

Thanks for finding this.

I will get this merged to master.

--Jeremy

PR #1121 (2024-01-27): Issue#1120: Uncommented lines of code
Resolves #1120.
Lines of code to remove `actual.sjon` in the teardown of `testMain` were commented out, leaving the test file behind after the test. Uncommenting the lines corrects the issue.
closed 2024-01-27 by jeremiah-c-leary