jeremiah-c-leary/vhdl-style-guide #1122: Exception on valid VHDL " 'close_parenthesis' object has no attribute 'iId' " when using --all_phases

maltaisn, 2024-02-08:
**Describe the bug**
When using `--all_phases`, the program crashes on some valid VHDL file.

**To Reproduce**
Here's a minimal example to reproduce the exception:

###### bug.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
   port (
      x : out std_logic
   );
end bug;

architecture behavioral of bug is
begin

x <= (a) + b(c(d)(e));

end architecture;
```

Execute the following command:
```
vsg --all_phases -f bug.vhd
```
I'm using vsg v3.20.0

The stacktrace:
```text
Traceback (most recent call last):
  File "C:\Python312\Lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
                    ^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\vsg\apply_rules.py", line 131, in apply_rules
    oRules.check_rules(
  File "C:\Python312\Lib\site-packages\vsg\rule_list.py", line 228, in check_rules
    oRule.analyze(self.oVhdlFile)
  File "C:\Python312\Lib\site-packages\vsg\rule.py", line 142, in analyze
    lToi = self._get_tokens_of_interest(oFile)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\vsg\rules\multiline_array_alignment.py", line 58, in _get_tokens_of_interest
    aToi = remove_non_arrays(self.assignment_operator, aToi)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\vsg\rules\multiline_array_alignment.py", line 196, in remove_non_arrays
    if rules_utils.array_detected_after_assignment_operator(assignment_operator, oToi):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\vsg\rules\utils.py", line 416, in array_detected_after_assignment_operator
    if open_paren_after_assignment_operator_id_matches_last_paren_id(assignment_operator, lTokens):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\vsg\rules\utils.py", line 467, in open_paren_after_assignment_operator_id_matches_last_paren_id
    iCloseParenId = lTokens[iIndex].iId
                    ^^^^^^^^^^^^^^^^^^^
AttributeError: 'close_parenthesis' object has no attribute 'iId'
```

**Expected behavior**
No exception, this is valid VHDL.

jeremiah-c-leary, 2024-02-10:
Morning @maltaisn ,

I was able to replicate the issue on release 3.20.0, and it looks like it was fixed on release 3.21.0.

Could you upgrade to version 3.21.0 and double check the issue is resolved?

Thanks,

--Jeremy

maltaisn, 2024-02-12:
Yes it is resolved in 3.21.0

closed 2024-02-12 by maltaisn