jeremiah-c-leary/vhdl-style-guide #1123: Hang on malformed array range followed by unclosed port

maltaisn, 2024-02-08:
**Describe the bug**
The program hangs when parsing a port declaration with an unclosed parenthesis in an array range.

**To Reproduce**
###### bug.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
   port (
      data : in std_logic_vector(11 downto;  -- !!!
   -- !!!
end entity;
```
Execute the following command:
```
vsg -f bug.vhd --jobs 1
```
I used `--jobs 1` otherwise the program cannot be stopped with a keyboard interrupt (is this expected?).

**Expected behavior**
I'm not sure what's the expected behavior on malformed VHDL. I've seen crashes in some cases, but a hang is a bit more annoying.

jeremiah-c-leary, 2024-02-10:
Good Evening @maltaisn ,

I just pushed an update for this to the `issue-1123` branch.  I added an exception to catch this case an report an error message.  When you get a chance can you try it on your end and let me know what you think?

Thanks,

--Jeremy

maltaisn, 2024-02-12:
It still hangs. I'm doing:
```
bin/vsg -f bug.vhd --jobs 1
```
Is this ok?

jeremiah-c-leary, 2024-02-13:
Evening @maltaisn ,

You will need to pull down the `issue-1123` branch and then re-run the command.

Using your code snippet I get the following outp;ut:
```mono
$ vsg -f original.vhd --jobs 1

Error: Closing token not found while parsing Line 6

 6 |       data : in std_logic_vector(11 downto;  -- !!!
```

Let me know if you need any help with getting to the branch.

Regards,

--Jeremy

maltaisn, 2024-02-13:
It works. I had just mixed the two issues. Thank you for the quick fixes.

closed 2024-02-13 by maltaisn

jeremiah-c-leary, 2024-02-14:
Awesome,

I will get this merged to master.

--Jeremy

PR #1126 (2024-02-14): Issue#1123:  Added exception for malformed discrete_range.
Resolves #1123