jeremiah-c-leary/vhdl-style-guide #1124: port_026 missing closing parenthesis in split ports

maltaisn, 2024-02-08:
**Describe the bug**
The `port_026` rule misses a closing parenthesis in some circumstances.

**To Reproduce**
###### bug.vhd
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
   port (
      ain1, ain2 : in std_logic_vector(11 downto 0);
      out1, out2 : out std_logic_vector(12 downto 0));
end entity bug;
```
The following command is executed:
```
vsg -f bug.vhd --fix
```
The resulting file:
###### bug.vhd
```vhdl
library ieee;
  use ieee.std_logic_1164.all;

entity bug is
  port (
    ain1 : in    std_logic_vector(11 downto 0);
    ain2 : in    std_logic_vector(11 downto 0);
    out1 : out   std_logic_vector(12 downto 0;    -- !!!
    out2 : out   std_logic_vector(12 downto 0)
  );
end entity bug;
```
I'm using vsg v3.20.0

**Expected behavior**
A parenthesis is missing.

jeremiah-c-leary, 2024-02-09:
Good Afternoon @maltaisn ,

Thank you for finding this issue.  I pushed an update to the `issue-1124` branch.  When you get a chance could you check it out on your end and let me know it works for you?

Thanks,

--Jeremy

maltaisn, 2024-02-13:
It works.

jeremiah-c-leary, 2024-02-14:
Awesome, merging to master.

--Jeremy

PR #1125 (2024-02-14): Issue#1124: Rule port_026 would miss closing parenthesis.
Resolves #1124