jeremiah-c-leary/vhdl-style-guide #1127: Code block is not showing on Configuring Multiline Indent Rules page

JHertz5, 2024-02-14:
**Environment**
```
$ ./bin/vsg -v
VHDL Style Guide (VSG) version: 3.21.0.dev8
Git commit SHA: 4c3164e7
```

**Describe the bug**
The code block is not showing under the heading "Example: align_left set to yes and align_paren set to yes" on the page [Configuring Multiline Indent Rules](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html#configuring-multiline-indent-rules).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to [this link'](https://vhdl-style-guide.readthedocs.io/en/latest/configuring_multiline_indent_rules.html#example-align-left-set-to-default-yes-and-align-paren-set-to-default-yes)
2. Observe that no code block appears under the linked heading.

This can also be observed when building the documentation locally.

**Expected behavior**
A code block should be present.

**Screenshots**
[image]

JHertz5, 2024-02-14:
I have raised PR #1128 to fix this.

jeremiah-c-leary, 2024-02-16:
Evening @JHertz5 ,

Thanks for catching that.  I will get it merged to master.

--Jeremy

PR #1128 (2024-02-16): Issue#1127: Corrected formatting for code block in "Configuring Multiline Indent Rules" page.
Resolves #1127.
closed 2024-02-16 by jeremiah-c-leary